josefadamcik/SofleKeyboard #1: rgb matrix

zhangzc, 2020-05-13:
Hi Josefadamcik,
Thanks for your great project.
I had changed pcb and added RGB led(sk6812mini) for all switch.
This is my first time using QMK, so I don't think I could modified the firmware.
May I ask for more information about adding rgb matrix in building qmk firmware ？
Thank you very much

josefadamcik, 2020-05-13:
Hi, that sounds cool.
But sadly I cannot help you with rgb matrix in QMK, I have never used it.
What I would do If I had to solve the problem:

- read the documentation
- find a split board that uses rgb matrix and check it's firmware
- ask on QMK discord

Good luck,
Josef

zhangzc, 2020-05-13:
Ok,Got it.
It seems I should go through the documents of QMK  and I will go to check the crkbd's firmware.
Anyway, thank you.

closed 2020-05-14 by zhangzc