josefadamcik/SofleKeyboard #2: Tenting?

AlexDaniel, 2020-06-22:
What about tenting? Maybe some extra holes should be included to make some basic tenting possible (using bolts with something rubbery on the ends for adjusting the height, like [here](https://i.imgur.com/5GDRDVC.jpg)).

josefadamcik, 2020-06-22:
@AlexDaniel thanks for the image, that's actually quite interesting solution. 

My plan was to explore tenting in the future and have a special version of case. Usually I have seen tenting support being done with "notches" which affected the shape of the case (like here: https://imkulio.com/build-guide.html) which is solution I wouldn't like as the default one since not everyone cares about tenting (I personally don't know yet if I care about tenting). 

But the image you attached is interesting, something like that could work. I'll think about it.

AlexDaniel, 2020-06-22:
@josefadamcik another interesting idea is something like this: https://imgur.com/a/0Gi4lRP

Basically, the bottom plate is attached with standoffs of different size, plus there's a difference in rubber pads. Makes the keyboard slightly angled, which can be enough for some users.

I'm also not sure about tenting in general, but when building a keyboard I'd like to have an option to experiment with it.

closed 2020-10-25 by josefadamcik