josefadamcik/SofleKeyboard #3: Inverted silkscreen?

AlexDaniel, 2020-06-22:
What about inverting the silkscreen on most of the PCBs? Like [here](http://ncmachineworks.ca/wp-content/uploads/2014/01/PC280491.jpg). It adds an extra layer and makes PCBs feel nicer. Also, with some fabs there is a cost for non-green PCBs, but the silkscreen fill esentially makes them white. I'm personally interested in black boards, but black soldermask from JLCPCB feels junky so I want to try my luck with white PCBs and inverted silkscreen (which should be black).

josefadamcik, 2020-06-22:
@AlexDaniel it might be interesting visually, but is there a way how to do it easily without loosing the ability to change the PCB design easily? Personally I have never tried to invert silkscreen so I don't know what are the options.

For me it sounds more like some kind of post-processing step: take exported Gerbers and run both silkscreen layers through some kind of inverting script.

AlexDaniel, 2020-06-22:
> easily

Well, maybe not as easy. But you can have text and other stuff on ECO layers, and then subtract it in gerbers. Maybe [this](https://forum.kicad.info/t/negative-silkscreen/11849/3) is a good start.

Another option is to have silkscreen artwork in SVG and then just import it into KiCad. It requires an extra step to bring in the updates but it's not too bad.

josefadamcik, 2020-06-22:
> > easily
> 
> Well, maybe not as easy. But you can have text and other stuff on ECO layers, and then subtract it in gerbers. Maybe [this](https://forum.kicad.info/t/negative-silkscreen/11849/3) is a good start.

Thanks for the link, it's interesting. But sadly not really practical way. The text is not the only thing on silkscreen. There are also symbols which are part of the footprints and that would be the more difficult part I would say. And they even have to use external script: gerber_combine -> i think something like that is more what you need - have a way how to say negate my silkscreen (which is something like substract it from something which is completely filled in so it might be possible to abuse gerber_combine for that purpose). 

>
> Another option is to have silkscreen artwork in SVG and then just import it into KiCad. It requires an extra step to bring in the updates but it's not too bad.

I would do that if there actually was some artwork I would care about. I have done some things  by combining SVG made in inkscape and KiCad (and importing/exporting things in both directions) ant it is indeed powerful approach. But it's also a lot of overhead. Doing it just to invert silk screen is too "expensive". I would prefer automated way that allows me to keep the project maintainable.

Also, I am quite sure it doesn't really make sense to have this as default - all my Sofle builds are made with white soldermask and black silkscreen and they look pretty good. I had Corne made with black-matte silkscreen from JLCPCB and it looked also very good (the only problem I had was that it was almost impossible to clean it nicely after soldering, at least I didn't find a reasonable way). Of course, this is highly subjective but there are plenty of people who might not like inverted silkscreen.

What I am trying to say here is: of course I am sure it's possible to do and of course I would be able to figure out why if I really wanted to. But for me this is something that you could do before you order PCBs yourself and doesn't have to be part of the repository. And if it was part of the repository it should a an optional post-processing step - e.g. a python script which would process the gerbers to make the change.

AlexDaniel, 2020-06-22:
> There are also symbols which are part of the footprints and that would be the more difficult part I would say

Any reason you can't move them to another layer? If it's just text, then right click on the text, Properties, Layer → Eco1.user. If it's part of the footprint, then Ctrl+E to edit it in the footprint editor.

However, I agree, it should be other way around. We should draw the silkscreen fill on the eco layer and then subtract the normal silkscreen from it.

> all my Sofle builds are made with white soldermask and black silkscreen and they look pretty good

That's alright aesthetically, but white color is perhaps the least functional for keyboards because of the grime and stuff.

Anyway I'll take a look on how to achieve it in the easiest way before making my order, and I'll submit a PR with instructions or scripts.

AlexDaniel, 2020-08-10:
If anybody is interested, I made an order for white boards with inverted silkscreen. I'll describe the process in a bit more detail later if the boards are fine.

joaommartins, 2020-08-24:
> If anybody is interested, I made an order for white boards with inverted black silkscreen. I'll describe the process in a bit more detail later if the boards are fine.

I'm interested since I intend to order from JLCPCB and it's my first time ordering pcbs at all.

AlexDaniel, 2020-08-24:
@joaommartins I hope you can wait a little bit. My boards were manufactured on 2020-08-16 but I chose the cheapest delivery option. The boards have various improvements to them, not just inverted silkscreen (something I intend to report as bugs soon). That said, I won't be sharing any files until I actually get the physical boards. I looked at the production files and JLCPCB did some undesired changes to the gerbers, especially the silkscreen. When I asked them not to do it they didn't really understand what I was trying to say… So I have to see how the boards turned out before I can recommend anything :)

joaommartins, 2020-08-24:
@AlexDaniel Not a problem at all, I'll be keeping an eye on this repo and learning about pcb manufacture meanwhile.

AlexDaniel, 2020-08-27:
They arrived! Looks pretty good to me, I'll explain the process in detail a bit later and also show some photos.

AlexDaniel, 2020-08-28:
OK, here are most of the details for making boards with inverted silkscreen.

Let's say you wanted to have a solid board (like the bottom plate) with silkscreen across the whole board, with no markings or logos (just solid silkscreen fill). Can you just cover the whole board with a polygon on the silkscreen layer? Like this:

[image]

The answer is yes, but with two caveats.

### 1. Pads
If you have any pads, most manufacturers will remove silkscreen from them. However, it is not guaranteed, as demonstrated by this famous case:
[image]
([source](https://www.eevblog.com/forum/projects/silk-screen-on-exposed-pcb-pads/))

KiCad allows to “exclude pads from silkscreen” when you generate the gerbers, and you probably want to make sure that the checkbox is checked:
[image]
There is also a “Subtract soldermask from silkscreen” option but I personally think it is too much. It will create holes for pads that are even bigger, but due to silkscreen misalignment it can expose some of soldermask too.

### 2. Edge cuts 
If you make your silkscreen exactly the same size as your board, keep in mind that the silkscreen is the most imprecise part of your boards. Usually, the silkscreen layer will be misaligned at least a bit, if not a lot. This means that your silkscreen polygon will be shifted, revealing soldermask on the edges of your board. Here's an example:
[image]
(boards for [one of my older projects](https://progarm.org/))

On the image above you can see green soldermask being visible on the left side but not on the right side, that is because of the silkscreen misalignment. The solution to that is to make sure your silkscreen overlaps the board edge, this way the mill will go through your silkscreen when cutting the board edge leaving no visible soldermask. However, this didn't really work for my JLCPCB order, and this is what they did for the production files:
[image]
(This is the screenshot that I was using when I was trying to explain the issue to JLCPCB, they still didn't get it though. The image on the right is the original gerber file, the image on the left is their production files. The arrow shows what you should want your gerber files to be, but the sad reality is backwards :) )

Notice that not only did they cut the silkscreen layer based on the outline, but they also made sure that it is inset by a small amount. That is, when ordering from JLCPCB, *you will always have a soldermask outline near all of the board edges*. This is how it looks like:
[image]

Notice that there is a white outline near all of the board edges.

When I asked JLCPCB not to do that they didn't really understand my request. I think it's much easier to embrace this limitation – just add an intentional 1mm-1.5mm silkscreen outline for all of your edge cuts. It will add contrast, making the board look more stylish.

### How to invert the silkscreen

Now, the process for inverting the silkscreen can be different depending on what exactly you want to achieve. For the top and bottom plates it is sufficient to simply add a polygon on silkscreen layers that covers the whole board. This is what I did, because I didn't want to have any graphics there anyway. You may choose to do it differently.

For the main board, however, I still wanted to have normal silkscreen markings for the components. So here is the process:
1. Generate gerbers with your normal silkscreen. Rename your silkscreen gerber files so that you know that this is the original silkscreen.
2. Add a polygon to silkscreen layers that covers the whole board. Generate the gerbers again.
3. Download [kicad-utils](https://github.com/bobc/kicad-utils) and use the [gerber_combine.py](https://github.com/bobc/kicad-utils/blob/v5/scripts/gerber_combine/gerber_combine.py) script to subtract one gerber file from the other. The result would be your final gerber file. Repeat for both layers! Unfortunately, `gerber_combine.py` script is somewhat broken, so you might need to modify it to get it working.

I went through a slightly more difficult process because I wanted to have “JLCJLCJLCJLC” string on the board. I don't know if they are able to invert the board number, so I did the following
1. I added a solid rectangle polygon to the normal silkscreen layer. This will later form a hole for our silkscreen text. Then I generated the gerbers like normally.
2. Same as described previously for Step 2, I generated silkscreen layers that cover the whole board.
3. I added “JLCJLCJLCJLC” string to one of the unused layers (in this case Margin layer). You can do this step later or ahead of time, it doesn't matter that much. The result should look like this:
[image]
4. Take your filled gerber file and *subtract* normal silkscreen from it, then *add* the Margin layer. It may sound complicated but it is actually very easy once you understand the order of operations. Your silkscreen will then look like this:
[image]
And, once manufactured, like this:
[image]

### Some other interesting details

JLCPCB will drill into the edges of square holes like this:
[image]

It makes sense because the inner corners can't be square, but notice that it also cuts into the copper and silkscreen around the hole and adds a 0.02mm artefact. I can't see it with my eyes so it's nothing to worry about, but maybe it means that the gerbers should include drill points on the corners right from the start.

JLCPCB also modifies overlapping holes so that one of them is milled like an edge cut instead:
[image]

[... cut ...]
* [Silkscreen error (R1/R2)](https://github.com/josefadamcik/SofleKeyboard/issues/5)
* [Missing silkscreen markings](https://github.com/josefadamcik/SofleKeyboard/issues/6)
* [Acute angles on copper layers](https://github.com/josefadamcik/SofleKeyboard/issues/8)

I think that's about it. Please let me know if there's anything I can help with.

Please feel free to close this issue because I solved the problem for myself. :) I hope my instructions are detailed enough for others to achieve something similar.

josefadamcik, 2020-09-06:
@AlexDaniel Thank you very much for taking the time to write this down and also to create issues for all the problems with the design you have noticed. I think I'll create a small section in the documentation (something like "Modifications") and link this issue & your comment from it so people can find it if they prefer inverted silk-screen. 

To the bugs: I'll leave them (at least most of them) here among issues open, but I am most likely not going to fix them in the first version of the design. But it's great feedback and I'll make sure the next versions doesn't have the same problems (at least in the cases where I would be able to find the solution). Thanks again.

AlexDaniel, 2020-09-06:
@josefadamcik thank you for a great project! The documentation is very nice!

closed 2020-12-21 by josefadamcik

olDirdey, 2024-11-05:
@AlexDaniel can you explain what you have changed in the gerber combine scrypt? in my output the silkscreen looks exactly the same as before, nothing is deducted after the script has run

AlexDaniel, 2024-11-05:
@olDirdey if I remember right, there were some obvious python errors, probably because the script is rather old. The script is written in python2 which isn't even easy to install nowadays. But, if you're able to run it without errors, then you're good!

How are you running it? Are you providing two files? Are you using the `-s` flag?

olDirdey, 2024-11-05:
@AlexDaniel thanks for the feedback. unfortunately I have no knowledge of python to find an obvious error. I use the command "python gerber_combine.py full.gbr -s text.gbr -o final.gbr" full is my filled area and text.gbr is the text i want to subtract from it. but the output file is the same as the file with the full area

AlexDaniel, 2024-11-05:
Based on the command history, I see that I was using this command:
```
./gerber_combine.py -s silkscreen-magic/SofleKeyboard-B_SilkS_FULL.gbo silkscreen-magic/SofleKeyboard-B_SilkS.gbo
```
I don't know if the order of arguments is important, but that's something to try. If you don't see any errors, then I think the script is fine.

olDirdey, 2024-11-05:
thanks again. I will try a little bit