josefadamcik/SofleKeyboard #10: Somewhat misleading surface finish instructions (lead toxicity)

AlexDaniel, 2020-08-28:
The website says this:

> Manufacturing […] Surface finish (keep in mind that the cheapest HASL contains lead. Lead is toxic.) and solder-mask color according to your taste.

I think this note can mislead people to choose a surface finish that they don't need. It is true that lead is toxic when ingested, but most people use leaded solder! So if one chooses ENIG but goes to use leaded solder then it defeats the purpose. It can be even more problematic if leadfree HASL is chosen. Of course, leadfree solder exists and can be easily purchased, but it is harder to solder with so most people don't use it.

Another thing is that the only way lead can get into your body is if you keep touching lead and then putting your hands into your mouth (lead cannot be absorbed through the skin). Given that top and bottom plates cover the main pcb, it is unlikely to happen. That being said, the exposed Arduino pins can be a problem.

Also, as I understand it, the biggest issue with leaded solder is that electronic waste is often dumped in massive amounts on/under the ground, meaning that lead can get into the ground water. It is not that big of a concern for individuals. Just don't eat it :)

Perhaps change the note a bit to recommend using lead-free solder with either lead-free HASL or ENIG.

josefadamcik, 2020-09-06:
AFAIK, It's actually illegal to use leaded solder in EU for anything else than repairs. Selling and import of leaded solder is becoming more and more regulated and it AFAIK shouldn't be sold to individuals (may differ slightly per country) Of course, as a hobbyist, one can easily continue to use it since nobody will ever really go after that. But it's not something I would like to encourage in the documentation. That's why I mention people should use a lead free variant. If they have some knowledge, they can decide for themselves. If they don't have the knowledge, they shouldn't use it.

AlexDaniel, 2020-09-06:
@josefadamcik but then you should also recommend to use lead-free solder and a good soldering iron. Otherwise people might try using leaded solder (without realizing) on top of the lead-free finish. That was my point.

con-f-use, 2020-09-10:
The main reason, we use lead free solder today is for environmental reasons. However, even that is heavily debated, because manufacturing with lead free solder uses more aggressive flux, that might potentially be more dangerous to worker health and the environment. Additionally, failure rates with lead free may be higher, leading to more waste.

AlexDaniel, 2020-09-10:
> Bottom line, it doubt have or not having the comment is going to make much difference

Well, I think the guide is meant to be understandable even for people who have little experience in the field. If it discourages leaded HASL, but doesn't say that then you should use lead-free solder, then at the very least some people can end up overpaying a bit. That's the point of this ticket.

AlexDaniel, 2020-09-29:
Actually, I'm changing my mind on this. When it comes to hobbyists like us, I think there's no good reason for anyone to use solder with lead nowadays. Especially given the open design of the keyboard and how close the arduino is to the keys. Nothing a nail polish trick can't solve, but no good reason to start with leaded finish and solder in the first place.

However, my point stands. The guide should also say that one should use use lead-free solder when working on the board. Yes, it is kinda obvious, but hey, maybe it'll help some beginner.

josefadamcik, 2020-09-29:
> However, my point stands. The guide should also say that one should use use lead-free solder when working on the board. Yes, it is kinda obvious, but hey, maybe it'll help some beginner.

You are right, I'll improve the forumlation in the documentation. Thanks for the feedback!

con-f-use, 2020-09-29:
What is the "nail polish trick"?

AlexDaniel, 2020-09-29:
@con-f-use a way of protecting pcbs with plain transparent nail polish ([example](https://youtu.be/h8ycmroFQSs?t=15)). Very cheap, widely available and it's probably the least toxic stuff for coating your pcbs. Because it is transparent, the PCBs remain serviceable but are much less prone to shorting (e.g. you dropping a coin on it, or something). Not sure how well it works for long-term waterproofing, but it does help for sure. As for the lead, no idea if it makes a difference, but I guess it should given that it stops you from touching the solder directly.

closed 2020-12-21 by josefadamcik