josefadamcik/SofleKeyboard #9: Impossible holes

AlexDaniel, 2020-08-28:
There are some holes that are too close to each other, they are almost impossible to be manufactured in a straightforward way.

Example:
[image]

I checked the production files from JLCPCB and it seems like one of the holes is drilled while the other one is milled. That said, the result is still far from being good:

[image]

Notice how sometimes there's a bridge between the holes that didn't break away at all. Other times the board has a bit of whiskers left over.

josefadamcik, 2020-09-06:
As many others this is inherited issue from Lilly58 pro. I have no Idea how this could be fixed while still having a double sided PCB an support fox both MX and Choc switch sockets. 
 
So far I have dropped support for Choc from the next version, but I think the socket making the most trouble here is the MX one so no clean solution there. It might happen that some future version will drop the double-sided pcb concept (once I move controller to the PCB it might be handy to have a 2 different PCBs). That would solve the problem for sure.

AlexDaniel, 2020-09-06:
> So far I have dropped support for Choc from the next version, but I think the socket making the most trouble here is the MX one so no clean solution there. It might happen that some future version [...]

Actually, both of these things is what made it a no-brainer for me to choose Sofle. I'm relatively sure that my next keyboard (if I ever make one) will use Choc switches.

One way of fixing the issue is to use the outline layer for both of the holes. For example, JLCPCB already moves one of the holes to the outline layer, so I doubt there's any disadvantage to move both of them. But there is an advantage, because the mill will cut both holes in one continuous move.

josefadamcik, 2020-09-07:
@AlexDaniel 

> Actually, both of these things is what made it a no-brainer for me to choose Sofle. I'm relatively sure that my next keyboard (if I ever make one) will use Choc switches.

Good to know. To be honest I would be the happiest if I had some kind of script that would fill in footprints for switches (type, socket yes/no, rgb led yes/no) and routing for the board so it could be configured and generated as anyone wishes. Everybody has a bit different preferences.

But I don't believe I would ever have time for something like that. 

> One way of fixing the issue is to use the edge cut layer for both of the holes. For example, JLCPCB already moves one of the holes to the edge cut layer, so I doubt there's any disadvantage to move both of them. But there is an advantage, because the mill will cut both holes in one continuous move.

Good Idea, I have noted it down and try to experiment with this in the future.

AlexDaniel, 2020-09-07:
Just to illustrate, here's a screenshot of jlcpcb prod files:

[image]

I wonder if kicad footprints can include edge cut drawings. If so, then it's probably the way to do it.

> To be honest I would be the happiest if I had some kind of script that would fill in footprints for switches

Interesting idea. Now that I think about it, it is actually not as hard as it sounds. I [have experimented](https://github.com/RGVID-EU/Butterboard) with generating entire PCBs via kicad api, but that is not needed in this case. Basically, what's needed is a slightly bigger generic footprint that has pads where the traces would connect. Then, each socket-specific footprint can include the required holes, pads *and also* traces that connect its pads to the pads of a generic footprint. This way you'd be able to swap socket types without having to reroute any traces on the board.

closed 2020-12-21 by josefadamcik