josefadamcik/SofleKeyboard #13: The outer bottom notch is wasted space that could fit a key

con-f-use, 2020-09-10:
[image]

Awesome work. My daily driver is an Iris rev. 4, and the only two things I was missing from it were 1) OLEDs and 2) about four buttons in the bottom row. So good job anticipating my need, I'm sure that is what you set out to do :sweat_smile: 

As to the issue, I was wondering if there is a particular reason why there is no 6th key in the bottom row (location marked red in above's picture). I feel that the notch there is just wastes space, especially because a key in that position would be more easily reachable than the <kbd>And</kbd> key in the picture. Also, it would help people that have the usual <kbd>Ctrl</kbd> key location still in their muscle memory. It could be notched from the other keys on the PCB, so that people who don't want it can remove it. I would really love that.

josefadamcik, 2020-09-10:
Hi @con-f-use thank you for your feedback! I am glad you like the design.

> As to the issue, I was wondering if there is a particular reason why there is no 6th key in the bottom row (location marked red in above's picture). I feel that the notch there is just wastes space,...

This is given by the fact how and why I created the design. I described my thinking here:  https://josef-adamcik.cz/electronics/in-search-of-the-best-custom-keyboard-layout.html

TLDR: I almost made Lilly58 but based on my habits I knew I would need one more key in the bottom row. But not more (again based on how I was using my Katana60).  And adding one key was less work. The first version of Sofle is really just a modified Lilly58. It's not designed from scratch. 

Also, I had an experience with Corne which was disappointed because I loved the minimal design in theory, but it was just too minimal for me. But I still had the mindset of having rather less than more. 

Sofle was never supposed to be a design for other people but rather a design for me and I the main point I wanted to illustrate when I published the desing and wrote about it and shared it was that it's not that hard to modify an existing open source keyboard design so it suits your needs. Even the name and the logo were more jokes and anything else.

Basically almost everyone who likes Sofle says: it's **almost** perfect for me, **but** there could be.. one key less, one key more, no encoders, a key instead of encoder... etc. If I wanted to take all the wishes and suggestions it would be an impossible keyboard. And in the end, that's why we have so many designs in the community.

> More easily reachable than the And key in the picture

Assuming you would be using pinkie but I don't do that. I even moved the bottom row further into middle so  the "and" key on the picture is now aligned with the 4th column. And 2 outer columns are lowered so they are easier to reach. (A bit about the current iteration here: https://twitter.com/josefadamcik/status/1271790779858640896?s=20 ) 

> Helps people that have the usual Ctrl key location still in their muscle memory

Muscle memory is very individual, this really depends on your keyboard and layout. And when you move from "traditional" keyboard to ergo split, I'll always hurt. Also I think, even with all the keyboards on laptops and similar, it's no longer really usual ctrl location.

> Many people like to press these kind of corner keys with the side of their hands

Well, good to know. I had no idea.

con-f-use, 2020-09-10:
I understand, and the beauty of open source, of course is, that people can fork a project and customize it to their needs.

Still, it would be an nice option to have that would be truely optional because of the notching. And if you didn't want people to like it and cater to their need, you'd not have made it public, so I don't buy it :stuck_out_tongue_winking_eye:

con-f-use, 2020-09-10:
> Muscle memory is very individual

In this case it's almost every keyboard user. Because most of them use a standard ISO or ANSI layout.

Thanks for the twitter link by the way, I enjoy pretty pictures.

closed 2020-12-21 by josefadamcik