josefadamcik/SofleKeyboard #15: Screw clearance with the encoder (and choosing the knobs)

AlexDaniel, 2020-09-29:
When clicking on the encoder:

[image]

Whoops. The screw position should be changed so that there is no chance for it to interfere with the encoder.

Moreover, the docs suggest these parts:
* https://www.aliexpress.com/item/32222859399.html
* https://www.aliexpress.com/item/32382989585.html

The seller is very nice. I asked them to put 5 knobs because the other lot has 5 encoders, and they did that with no extra charge. However, that combo just doesn't work for Sofle. Even on the photo you can see that the encoder sits a bit too deep in the knob, so depending on tolerance some knobs will bottom out against the PCB itself. A different knob should be found.

josefadamcik, 2020-09-29:
```
Whoops. The screw position should be changed so that there is no chance for it to interfere with the encoder.

Moreover, the docs suggest these parts:

    https://www.aliexpress.com/item/32222859399.html
    https://www.aliexpress.com/item/32382989585.html

```
I feel sorry about this, you are of course right. The suggestions are not the best ones. It's hard to find a good knob on aliexpress and I should have updated the suggestion to the ones I actually use. Thanks for the feedback, I'll fix this ASAP.

AlexDaniel, 2020-09-29:
No worries. Also, forgot to mention, this is “fixable” by dropping an M3 nut into the knob. It doesn't fix it completely, but at least it is now usable while I'll be waiting for the right knobs.

closed 2020-12-21 by josefadamcik