josefadamcik/SofleKeyboard #22: Use PD4 for row0 for matrix rather than PC6

josefadamcik, 2021-01-05:
The pins are next to each other and PD4 is currently unused. But PC6 (which is used for row0) supports PWM output (Timer3 ouput) so it might be usefull if somebody want's to add a speaker.

It might be also nice to have some pad for this pin somwhere in a similar way as for RGB pin PD3.

Datasheet: https://cdn-shop.adafruit.com/datasheets/atmel-7766-8-bit-avr-atmega16u4-32u4_datasheet.pdf

harshitgoel96, 2021-03-25:
Hi. Let me know if you need help on this. I can make the change and submit a PR.

josefadamcik, 2021-03-25:
@harshitgoel96 
> Hi. Let me know if you need help on this. I can make the change and submit a PR.

Thank you but no, not really. This is technically a breaking change (would brake firmware) so it's not going to happen for any of the current versions, it's waiting for potential version 3.