josefadamcik/SofleKeyboard #25: Right side of keyboard not staying on

AlexEales, 2021-01-09:
Not sure if this is the correct place to put this issue but couldn't think of anywhere else to ask!

Just put together both halfs of my keyboard and loving it so far! However, I have flashed the left side fine and all keys work but when coming to flash the right side I plug the board in and the light on the Bit-C controller comes on for 3-5 seconds before turning off again. This is also reflected in the output from the QMK toolbox:

```
*** Atmel DFU device connected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)
*** Atmel DFU device disconnected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)
```

I've gonethrough and checked my solders and nothing seems to be out of the ordinary so I wasn't sure if there is a step missing in the docs or if there is something wrong on my end I'm unaware of?

Thanks in advance!

AlexEales, 2021-01-09:
Update: the left side has now also stopped being detected

For reference this is on Windows 10 and also tried on my Ubuntu laptop

josefadamcik, 2021-01-09:
@AlexEales Hi, sorry to hear you are having trouble. Haven't seen similar problems myself, so I don't really know. But my guess would be some HW issue (short circuit, problems with micro controllers etc).

AlexEales, 2021-01-09:
Hi! Thanks for the speedy response everything has been going smoothly until now! Sorry to trouble but is there any resources on dealing with a SC microcontroller etc.?

closed 2021-01-09 by AlexEales