josefadamcik/SofleKeyboard #28: Although the sofle is "based on" the lily58, I can't use the wpm animation/wpm counter function that the lily58 can able to use.

thienanbeou, 2021-01-23:
I copied some WPM-responsive animation codes from a lily keymap, and qmk MSYS just didn't accept it.  
Can somebody help?
This is the source where I copied the codes, and yes, those codes did work perfectly on the lily: 
https://github.com/tnhh00037/qmk-lily.git

josefadamcik, 2021-01-25:
Hi @thienanbeou 
I would suggest (in general) to direct questions about QMK firmware towards QMK community. They have very active discord full of ppl who are much better at details of QMK than myself and you'll be more likely to find someone who has time and knowledge to help you. (As example, I have no clue what "qmk MSYS just didn't accept it" means and how could I help you with that detail, I would be wildly guessing).

I gave a quick glance to the keymap you have linked and it seems to by just copy&pasted from Lilly58 without any change (the layout is not Sofle). This cannot work. The layout is different -> the matrix is different -> you cannot use keymap from a different keyboard since it will not even compile. I would suggest to take the default Sofle layout and integrate the related code you into that.

Good luck

thienanbeou, 2021-01-25:
Thank you very much for the reply!

Actually, I did not copy the whole keymap of the lily, just only the parts
that are related to the "bongo cat animation". With the help of a friend of
mine and the splitkb.com Discord community, I've be able to work the code
out flawlessly just a couple of hours ago. Turns out that I was missing a
couple of commands, and I just need to change things a bit. Sorry if my
comment on Github bothers you somehow...
Here is the picture of my Sofle right now.
And, if you like, here's my keymap.
https://drive.google.com/file/d/1XPWZWC77yZbe0sZS0K4uwysL8MLZY9FW/view?usp=sharing
Thanks a lot for your reply Josef! I do really appreciate it!

josefadamcik, 2021-01-25:
@thienanbeou No problem, happy you figured that out!

closed 2021-01-25 by josefadamcik