josefadamcik/SofleKeyboard #32: Led type?

kimipsen, 2021-02-03:
I am having trouble locating the leds mentioned in the rgb guide (sk2812). Is it supposed to be sk6812 or ws2812(b)? Or am I just having the worst luck finding these leds?

This might be a question for @DaneEvans I guess.

DaneEvans, 2021-02-03:
Sorry, it's supposed to be sk6812 mini 

Fixing it up now

josefadamcik, 2021-02-03:
@DaneEvans Thank's a lot for quick reaction!

@kimipsen Pay attention to the "mini" part, they are small ones 3.5mm (the package is called 3535).

Adafruit is selling them as their "neopixel mini": https://www.adafruit.com/product/2686 digikey has them too: https://www.digikey.com/en/products/detail/adafruit-industries-llc/2686/5804107

kimipsen, 2021-02-03:
@DaneEvans @josefadamcik Perfect! Thanks for the great help here :)

closed 2021-02-03 by kimipsen