josefadamcik/SofleKeyboard #34: Wrong hole?

BeTmAsTeR, 2021-02-04:
I'm know nothing about pcb design so i'm just reporting an issued that JLCPCB faced and kindly reported back to me.

I sent JLCPCB the gerber files for v2 and they sent me this mail

> As shown below, do you need to cut out the circle at the arrow on PCB? It is in KO layer.
> 
> 
> Or we can delete it to proceed?
[image]

Is this an issue?

josefadamcik, 2021-02-04:
Hi Matteo, thanks for letting me know.

They can delete it and proceed.

I'll check the gerber's again to fix them, not really sure what happened
there.

Best regards,
Josef

josefadamcik, 2021-02-04:
Fixed int the KiCad files, gerber files and crated a new version with fixed files at https://github.com/josefadamcik/SofleKeyboard/releases/tag/v2.0.1

commit 3e2f8a1d (2021-02-04): Merge pull request #35 from josefadamcik/fix/34-v2-edge-cut-error

Fix #34
commit 9385cbae (2021-02-04): Fix #34 - a mistake in the edge-cut layer of version 2, gerbers updated too
closed 2021-02-04 by josefadamcik