josefadamcik/SofleKeyboard #37: Is there a raw data/json file I can import in to keyboard-layout-editor.com? I would love to use the layout.

mvaneijgen, 2021-02-07:
I'm in the process of researching some layouts for a keyboard I want to build. I would love to thinker with the v2 of this layout, to get a better feel for it!

josefadamcik, 2021-02-07:
There is layout file I have created some time ago for version 1:
http://www.keyboard-layout-editor.com/#/gists/76efb423a46cbbea75465cb468eef7ff
You should be able to fork it or download the json.
The v2 didn't really change much, the layout is practically the same only
some of the keys moved a bit. Nothing that would need to be reflected in
KLE. Adjusting ey positions etc. in KLE is not fun (in particular rotation
is meh).
Hope this helps.

mvaneijgen, 2021-02-07:
Thank you! [From your post](https://josefadamcik.github.io/SofleKeyboard/) it seems [a fair bit](https://josefadamcik.github.io/SofleKeyboard/images/Comparison1_2.png) had changed, but thanks for the reply and the layout, I'll be checking it out!

closed 2021-02-07 by mvaneijgen

mvaneijgen, 2021-02-08:
I think I've mostly recreate [the v2 layout](http://www.keyboard-layout-editor.com/#/gists/bac975add3bc3b9a76bd361f3ed9d003) in KLE. Hope it'll help someone else.

josefadamcik, 2021-02-08:
@mvaneijgen Thanks for sharing!