josefadamcik/SofleKeyboard #38: Sofle RGB has an unconnected via for NET34. Underlight LED circuit.

jmo808, 2021-02-10:
I temporarily worked around the issue by scratching the soldermask off and globbing some solder. I will submit a PR in the next few days.

DaneEvans, 2021-02-11:
That's a weird one.

Thanks for the quick turnaround, hopefully doesn't catch anyone else out, I
know there are a few people looking at making one.

I'll check it out tomorrow probably, if you haven't submitted one by then.

jmo808, 2021-02-11:
Here's some pics as an FYI. 
[image]
[image]
Second pic has where the missing trace was circled in red. It's an easy fix in KiCad

jmo808, 2021-02-16:
I finished building both sides of the keyboard and have verified that this fix does indeed work. So I will submit the PR soon.

tiadobatima, 2021-02-26:
Howdy... I'm wondering if this PR has been merged? I gave a quick look and didn't see any merges related to that in the last few days, but I could have missed something. I'm gonna wait for it before ordering :)
And thanks for the fix! ❤️

josefadamcik, 2021-02-26:
> Howdy... I'm wondering if this PR has been merged? I gave a quick look and didn't see any merges related to that in the last few days, but I could have missed something. I'm gonna wait for it before ordering :)
> And thanks for the fix! ❤️

Hi @tiadobatima, there has been no PR opened yet.

jmo808, 2021-02-26:
Sorry for the delay, I have gotten busy with work and haven't had the chance to submit this PR. I still need to load the JLCPCB DRC rules into KiCad and export the gerber. I'm also working on another PR for a 3d printed high profile case.

tiadobatima, 2021-02-26:
I'm so sorry... I didn't mean to put any pressure. I know everyone is busy. I just wanted to make sure I didn't miss the boat :)
Thank you again for doing this!

DaneEvans, 2021-02-27:
I'm in the same boat, work has been absolutely hectic.

I'm hoping to get one out this weekend. And update the documentation to
mention the issue in the R1 boards.

DaneEvans, 2021-02-27:
I've submitted a merge request with a new version of the board, and updated the documentation. 

@tiadobatima, wait a day or so for Josef to accept it, or you can pull it from my repo. 

Thanks jmo for finding (and immediately solving) the issue - My apologies for the hassle, I never made the 2.0 boards, they just fixed a few issues I had in 1.0, and somehow I missed a missing net in the DRC.  How I have no idea, because I now I ran the checks, and it showed up as son as I ran them to start the fix.

closed 2021-02-27 by josefadamcik