josefadamcik/SofleKeyboard #39: gateron switch not secured in top plate

sarlej, 2021-02-17:
Hi, i have problem with top plate pcb and gateron brown switches. Switches doesn't "click" in to position. They can be easily removed from plate and they swap out when i want to change keycap. I removed some material from bottom of top plates, where protrusion from switches are supposed to hold switches in position and now everything works well. I measured plates and they are 1.6mm as recommended in build guide and i ordered pcbs from jlcpcb with parameters from build guide (see details in picture).

Did i miss something or do something wrong?
Could it be problem with gateron switches?
Does anyone else have similar problem?

This was my first keyboard build, and except this issue i must say everything works flawlessly on first try. Thank you for great work.
Order details from jlcpcb (ENIG RoHS is crossed out out because i changed order later): 
[image]

josefadamcik, 2021-02-18:
Hi @sarlej, sorry to hear you are having trouble.

> Did i miss something or do something wrong?

Don't worry, you didn't. 

> Could it be problem with gateron switches?

Partly, maybe.

The design of the top plate for Sofle (actually for all DIY keyboards which use the top plate made from PCB) rely on the fact that the MX switch has a small notch to click on the plate which is almost the same dimension as the default thickness of PCB. This is a happy coincidence and not intended way how to mount the switches from the switch-manufacturer's point of view. Dimensions can differ slightly when it comes to switch clones, versions and revisions and I guess that manufacturing tolerances (for both switch and PCB) could mess things up. 

I would say there's nothing to do from my side to fix that. The only thing I can control is the size of the cutout and I am not sure if / how that could help in this case.

> Does anyone else have similar problem?

I guess I don't, but I must admit that my Zilents don't hold as firmly in the plate as I would like them too. They are also made by Gateron. But it's not that bad, at least I had no urge to fix it and I have no issues with my builds during daily use.

sarlej, 2021-02-27:
Thank you for information. Actually I'm not satisfied with Gateron brown, so i will try another switches. Hope they wont be too loose after i modify pcbs.

closed 2021-04-07 by DaneEvans