josefadamcik/SofleKeyboard #40: V2 Choc variant

brianlow, 2021-02-26:
I am working on variant that supports Kailh choc switches, optional hotswap sockets and per-key RGB. 

https://github.com/brianlow/SofleKeyboard/tree/choc/Sofle_Choc_v2

Welcome feedback and/or collaboration with anyone interested.

DaneEvans, 2021-04-07:
I'm happy to do reviews if you need before releasing gerbers

brianlow, 2021-06-11:
Update: one half is built and working. Everything is tested except for the encoder and TRRS connector. One trace is a bit too close to a pad and requires careful soldering but otherwise good

[image]

josefadamcik, 2021-06-19:
Well done, people keep asking about low profile Sofle V2 and this is even RGB. Once you have the final design, feel free to open a PR to this repository and we could talk about merging it. Of course only if you want. There is nothing wrong with having your own version / fork.

dattanchu, 2021-07-28:
@brianlow This is fantastic work. From the commit log, it looks like this support both mini-e and choc switch? Is the pcb files ready for physical production?

Can you elaborate on the hard-to-solder spot you mentioned please? My experience is limited to a non-led sofle RGB and would like to try my hands on a choc board for personal use.

brianlow, 2021-07-28:
> it looks like this support both mini-e and choc switch?

Yeah!

> Is the pcb files ready for physical production?

Short answer is not yet. Long answer: I am feeling pretty good about it. Most of the 2nd half is going. I've got a few keys/leds that probably need some soldering touch ups. I haven't tried connecting the halves. If you are adventurous I suspect it is ok to build. If you'd like I can make a few changes, like fix the hard to solder spot. 

> My experience is limited to a non-led sofle RGB 

You should have no trouble soldering this. I am a hobbist too. The SK6818 mini-e LEDs have been fairly easy. I've only over-heated or mis-soldered a couple leds.

brianlow, 2021-07-28:
Changes I was thinking
- [ ] inner thumb key, right side has a trace too close to the pad for the socket. I am surprised DRC didn't flag this. **Any suggested settings so this problem, and any others, are found automatically?**
[image]

- [ ] top plate: cutout for the encoder is just a 1-2mm too small vertically
- [ ] some of the text done with solder mask (?) isn't visible at least on white PCBs, change to silk screen

DaneEvans, 2021-07-28:
Did you run DRC?  For some reason kicad makes this easy to skip.

Check your rules too, but yeah, that should have been picked up I would
have thought.
Kicads DRC isn't the most intuitive, and a lot of things are turned off by
default.
Maybe increase the number too.

I'm a little surprised that the PCB house ran it.  Maybe it's not as bad as
it looks on the screen cap.

Text is usually either done in silk, copper, or mask over copper, mask
alone isn't common, bare FR4 isn't very attractive, and it only works with
coloured masks

brianlow, 2021-07-28:
@DaneEvans thanks! 

I was just experimenting with Footprint Editor -> select pad -> Properties -> Pad Clearance = 0.30 (default was 0). With this, the DRC finds this problem and a few others.

> I'm a little surprised that the PCB house ran it.  Maybe it's not as bad as it looks on the screen cap.

Here is a screen cap with measurement:
[image]

I ordered from JPCB. Maybe this is normal clearance and I was just too sloppy / too hot with soldering and lifted the soldermask or something.

DaneEvans, 2021-07-28:
That still sounds like a trace clearance number is too low, a pad clearance of 0 can't be normal, it has to be added to a trace one that's .3 or similar I would think.  

I do know I had similar issues where it allowed auto route to do things that seemed completely unreasonable. Thankfully I caught them.  

That one also looks to be inside the solder mask cutout!  
I think JLC charges more for them to check it, some of the others have a DRC tool on their upload site.

For a board like this, hand soldered by hobbyists, I aim for 2 layers, 
1+mm clearance, and more around pads.
Good gaps between diodes etc
And fairly wide traces, at least one to 2 sizes up from the default. 
Obviously that depends on density, but there is heaps of space on these

brianlow, 2021-07-28:
> a pad clearance of 0 can't be normal, it has to be added to a trace one that's .3 or similar I would think

Looks like 0 means it inherits "from the parent footprint" but I can't figure out from where. "Default Pad Properties..." seems to only affect newly drawn pads.

[image]

> For a board like this...

 That is great info, I'll try to incorporate

brianlow, 2021-07-31:
v2.0 hardware is working! 

I think safe to say ok to build this version. Happy to help any builders with gotches mentioned earlier. Will work on v2.1 PCB but no timeline. 

I've added a TODO list to the issue description. Will keep updated.

dattanchu, 2021-07-31:
Fantastic news. Given that I am working on my second sofle RGB board at the moment, I am gonna wait for the potential v2.1. For my shopping cart, I assume I can use the choc V2 switch as well?

brianlow, 2021-07-31:
[Kailh Choc v2 PG1350](https://www.aliexpress.com/item/32959996455.html) and will need matching [sockets](https://www.aliexpress.com/item/33023283633.html) (required)

josefadamcik, 2021-08-14:
Hi @brianlow, I am just putting together a yearly article "What's happened around Sofle in the last year" and I  am going to  mention  your modification. Can I use one of your photos from above in the article? And maybe later on the website (https://josefadamcik.github.io/SofleKeyboard/) as example of mods to the design? I'll credit you of course (and can link your website or something if you wish).

brianlow, 2021-08-14:
> Can I use one of your photos from above in the article?

Cool, yeah for sure. I am super close to opening a PR for this repo if still interested. Can probably open one this weekend.

blewisemt, 2021-11-08:
Super excited to have found this. Still trying to make heads or tails of the github and how to put one of these together, but this is precisely the keyboard I have been looking for.

lytedev, 2021-11-09:
@blewisemt I just finished my build. I have a number of extra PCBs if you want to avoid some of the process.

Download the gerber files and upload to JLCPCB (or wherever you want). https://github.com/brianlow/SofleKeyboard/tree/choc/Gerbers/Choc

brianlow, 2021-11-09:
@lytedev thanks, how did the build go, any problems? I think feedback would help with the PR. I'm also super curious to hear :)

blewisemt, 2021-11-09:
@lytedev if you happen to have some, that would definitely streamline the process. Having never attempted such a project before, how do I know what all I need?

brianlow, 2021-11-09:
@blewisemt 

> how do I know what all I need?

Go to PR #104 -> Files tab -> scroll down to `docs/build_guide_choc.md` to see the build guide. All the linked images are also in the PR. To view the guide as a formatted web page you'll need node and jekyll to compile it if I remember correctly.

This is a pretty rough project. Only 2 keyboard have been built as far as I know. If you want something a little more mature, check out the Lily58 and Iris. Both support choc, have kits available and good docs.

blewisemt, 2021-11-10:
Thanks for the feedback. I've ordered all of the components, as well as the
PCBs. Assuming my skills are up to the task, there should be a third
floating around soon! Anything in particular I should be on the lookout
for? I'm still working my way through the contents of the PR, and the PCB
design in particular.

lytedev, 2021-11-10:
@brianlow The build went great! I had absolutely no issues with the PCBs or plates. I would love to get into PCB CAD and figure out how to customize these things to my needs or even learn to make parametric PCBs to people could swap different components easily somehow.

My build is fully wireless using Nice!_Nano MCUs and ZMK-based firmware. I added hard power switches for even better battery life and the board has been wonderful.

I see no reason this shouldn't be merged, but again, not a PCB expert. Also just realized you included a build guide! Fantastic!

@blewisemt I was able to build it without even the build guide, but this is not my first build, so I was pretty familiar with the basics and the process as a whole. If you follow the build guide and make sure no power is running through anything while you're working (no batteries or cables connected) you will do just fine. It sounds like you already ordered PCBs so you don't need any from me!

If you have any issues, feel free to reach out!

Here's a pic of mine:

[image]

brianlow, 2021-11-10:
@blewisemt that was fast! Great to hear. Hopefully the build guide has everything you need. Ask if not and I can improve the guide. 

@lytedev great build, thanks for posting. Power switch on the pcb would be a great option

josefadamcik, 2021-11-14:
@lytedev Thanks a lot for building, testing and reporting back :)

closed 2021-11-14 by josefadamcik

blewisemt, 2021-11-20:
Hi all. Got my keyboard assembled. The build guide was extremely straightforward, and it functions *almost* perfectly. So where did I go wrong? I used the via setup, and all the keys work, but only the first vertical column on each half lights up. Thoughts?

brianlow, 2021-11-21:
@blewisemt 

Some ideas
* try the non-via firmware
* replace the first broken LED and the one immediately before it

For reference the LED configuration in `keyboards/sofle/keymaps/choc-brian (or choc)/config.h`. E.g. here is the via version:

https://github.com/brianlow/qmk_firmware/blob/5264ae0ce57033322638b87ada26a377a4b25eab/keyboards/sofle/keymaps/choc-brian/config.h#L17-L19

blewisemt, 2021-11-21:
So the first vertical column on each side works, and all of the LEDs test as functional. I wanted to run the non via firmware but I can’t seem to figure QMK out.

blewisemt, 2021-11-22:
This seemed to rectify my LED issues. However the OLEDs no longer display
anything and I cant seem to figure out how the layer up and layer down keys
work. Thanks for all the support thus far btw!

brianlow, 2021-11-22:
@blewisemt please submit a new issue on this repository (probably type Help Me) with details. This is a closed issue and posts here will notify everyone watching. What action fixed the leds? If you revert that action does the oled work again. What firmware are you running?

researcherx00, 2021-12-14:
@lytedev Still have any spare PCBs?
If so, can I get a set? 🤞 😄

brianlow, 2021-12-16:
@researcherx00 contact me at the email on my GitHub profile. Might be able to help

killua99, 2022-01-25:
@lytedev can I reach you out about ZMK firmware and dead OLED? thanks in advance and sorry for the mention

killua99, 2022-03-31:
@lytedev thanks, actually I got the OLED working, not the version RGB v2.1 is my next challenge, I still can not make LEDs to work. Using nice!nano v2 and ZMK firmware, since the Bluetooth support is the main feature I´m looking for.

If I made some advance on it perhaps I'll update this comment to don't cause more notification on old threats

brianlow, 2022-04-01:
@killua99 lets collect ZMK info in a dedicated issue since this one is closed: https://github.com/josefadamcik/SofleKeyboard/issues/142

PR #104 (2021-11-14): Sofle Choc variant
[image]
[image]

The Sofle Choc is a variation of the Sofle with:

   - low-profile, Kailh Choc switches
   - hotswap sockets
   - per-key RGB
   - thin, 15mm from desktop to top of keycaps
   - Sofle v2 switch layout with the more aggressive pinky stagger

I've built the v2.0 version and use it daily. This PR is a v2.1 version with a few minor fixes/tweaks but I have not built it. v.2.1 has been built by a few poeple so far.

Can preview the build guide in this PR here:
https://github.com/josefadamcik/SofleKeyboard/blob/27579fdc7e9a97dca35b73e7a9bfa7a9111230ac/docs/build_gu [... cut]