josefadamcik/SofleKeyboard #42: Question about LED soldering

armageddion, 2021-02-28:
This isn't an issue, so please close if inappropriate. 
 
Any tips on SMD soldering of the LEDs? I managed to get a few working on my Sofle RGB build, but i feel that it was more luck than skill. I used plenty of flux and minimal soldering temp and time. Here are a few things that I tried:
 
Method 1: I tried to apply a small amount of solder to each pad, place LED and blast it with a heat gun. That got the LED attached (on all 4 pads it seems) but the LED wouldn't work.

Method 2: Apply solder only to 1 pad, attach LED to that pad. Then feed solder to remaining 3 pads. 

Method 3: Apply solder on all 4 pads on the PCB, place LED on top. Apply low downward pressure on LED with tweezers. Melt solder on each pad until LED sitting comfortably in place. 

I suspect that I might fry the LED in method 1, but 2&3 shouldnt suffer from that. I am scouring the net for helpful tips on this, so I thought i'd ask here as well since I suspect I'm not the only one having this challenge.

DaneEvans, 2021-03-01:
None of those sound like a good idea to me.

The ideal would be solder paste, and an oven or wave solder rig.

The best luck I've had has been to flux the PCB,
then put the led on in the right orientation,
Hold it in place, while you use an iron with a blob of solder on it to heat
the PCB pad. Once it flows, touch the iron to the led and the edge of the
LED pad a couple of times.

It's a much easier job than the per key ones.   Bridging the pads without
frying the LED is awful. Particularly if your LEDs have any contamination
on the pads.

armageddion, 2021-03-01:
Thanks for the prompt response. 
Will try what you suggest. 

Side note: this is one epic keyboard and a fun build.

DaneEvans, 2021-03-01:
Haha, I try to be quick so that the RGB stuff isn't too much of a hassle to
Josef, given that he sensibly avoided them. They are a lot of work.

Also, did you have a V1.0 board? Have you seen issue #38?

sdkrocks, 2021-04-06:
> I did see that issue, but I didn't get that far yet. I am still on the Indicator LEDs.
Did you get anywhere?

I am having a similar issue. Can't get the indicator lights to work, have resoldered them with new leds a bunch of times at this point. Shows no signs of life.

DaneEvans, 2021-04-06:
Which version of the board do you have SDK? I know 1.0 and 2.0 people
succeeded with, except for issue #38, if you have a 2.1 it's possible but
unlikely for there to be a hardware bug.

If you have an RGB strip you can check the routing quite easily and
eliminate issues other than the LED orientation and soldering.

DaneEvans, 2021-04-07:
Added to build guide a few commits ago

closed 2021-04-07 by DaneEvans