josefadamcik/SofleKeyboard #45: SK6812-MINI-E

tiadobatima, 2021-03-03:
Hello there... Not sure this is wanted, but I'm starting the work to update the SofleRGB to use `SK6812-MINI-E` which seems to be considerably easier to solder vs the regular `SK6812-MINI`.

I have updated the schematics to use the `MINI-E`. I'm not good with Kicad, so I'll need some more time to update the footprint, but I think this is ready for early review if you're interested.

I couldn't find an official datasheet only but this is what some vendors on Aliexpress are publishing:
https://ae01.alicdn.com/kf/H79d59ee537124777a478665921acaa59A.jpg

I have the work on this branch on my fork:
https://github.com/tiadobatima/SofleKeyboard/tree/sk6812-mini-e

And to make your review a bit easier, I created a PR against my fork's master:
https://github.com/tiadobatima/SofleKeyboard/pull/1

Thanks! :)

DaneEvans, 2021-03-03:
> tiadobatima#1 <https://github.com/tiadobatima/SofleKeyboard/pull/1>

There should be no need to update the schematic, and it will only give more
places for things to go wrong.

Just create a new footprint, allocate it to the schematic parts and update
the PCB.

I'm happy to do a review and approve it,  but as I'm not going to update
mine, I'm not going to do the change myself, if the pinout matches it
shouldn't be too much work to route though.

tiadobatima, 2021-03-03:
Yeah... The pinout for the mini-e seems to be different than the mini, and that's the reason I thought the schematics needed to be changed, though I kept pretty much everything as is, apart from the mini-e itself and some minor cosmetic changes.
Did I understand you correctly that I could have kept the schematics the same even if the pinout is different? Mind sharing the trick?

Thank you very much @DaneEvans ! :)
g.

josefadamcik, 2021-03-03:
@tiadobatima 
The symbol in schematic should stay the same because electrically it's the same component.

Each symbol in schematics has assigned so called "footprint" which defines how the given component will look in the PCB design. 

Here's a bit more about how to change footprints: https://forum.kicad.info/t/how-can-i-assign-a-footprint-to-a-symbol/8901

tiadobatima, 2021-03-04:
Thank you for the reply guys! Yeah, I started playing around with the footprints yesterday, and in fact that was another reason I thought the changing the schematics would be the way to go, because unless I'm not looking into this the right way, if the component's pinout is different and we don't change the schematics they would be "documented" wrong in two places (in the symbol schematics and footprint)? What am I missing? I'm sure there's a logic behind it, but I didn't catch it yet :)
But yeah, I agree with "don't change if it works!" :) 

Anyways, the footprint brings me to a more important question: The current RGB LED is defined in two different footprints:
- K6812MINI_underglow_dev (the RGB by itself)
- SK6812MINI_and_cherry (RGB + switch)

Between these two footprints, the RGB pads in the copper layer and the cutout squares for the Edge.Cuts/Fab layer have different sizes. Is there a reason they're different because my first impression is that the smaller cutout and bigger pads would work better for the switch as well (lower probability for the LED to fall thru; and bigger pads would give slightly better heat dissipation and area when soldering).
I imagine it's because the the board can be used both sides, but I feel the smaller footprint would still work on switch and single footprint could be use for both?

Thank you again! This is a top notch open source project! 😃

josefadamcik, 2021-03-04:
@tiadobatima Just regarding the first bit sinc I don't have the knowledeg to answer the second question.

> I thought the changing the schematics would be the way to go, because unless I'm not looking into this the right way, if the component's pinout is different and we don't change the schematics they [...]

I think the the detail you might be missing is to fully understand what is the purpose of schematic and what it actually means that the component has a different "pinout".

The two LEDs are internally equivalent therefore they require to be connected in the same way - GND, PWR, data in, data out. That's the only point you care about in schematics and this haven't changed.

What changes is how are those connections placed in the real world when it comes to the particular instance of hardware (it's package). And to capture this we have footprints. 

For example, you can have a simple resistor: that's always a simple component with 2 connections where orientation doesn't matter. You'll have one symbol in schematics for it, but there is a whole range of packages which vary in size and the way how they are mounted to the PCB. So you have huge amount of footprints you can use when designing your PCB in order to support this variety. But the symbol in schematics is always the same.

DaneEvans, 2021-03-04:
The reason for two completely different footprints is that the underglow
one is a SMD footprint, it has no cutout, so the pads have to go under the
LED to work.

The combined one is  for the per key lighting, and has the cutout (although
I'm not sure I found a way to get it into the footprint in KiCad, it may
only be on a mechanical layer).

Any other sizes differences were because I stole footprints that worked
(crkbd I think) rather than doing them myself, especially for the glorious
hack that is putting LEDs on the back side of the board shining through. I
didn't want to do multiple runs to find a working footprint.

The combined footprint was my addition - I didn't want to have to line up
the LED each time, or have different part numbers for the key and
associated LED. I'm hoping it catches on, because apart from the creation
of the footprint, it's much easier to work with.

I need to relink my footprint library (thanks KiCad) and then I should be
able to just do a footprint swap to match the mini-e recommended footprint.
They seem close enough that it should only be a matter of pushing a few
traces if needed.

DaneEvans, 2021-03-08:
@tiadobatima 
Definitely don't do any more just yet. 
At most it's going to be a footprint replacement, at minimum it'll be a drop in replacement - 

I compared the two footprints, and they should be compatible (mini-e in green)
[image]

It won't be ideal, the pin may overhang the pad a little, but it should work as a drop in. I'll order some LEDs and try it out on one of my spare boards. (in whatever the ali-express lead time is) 

I'll probably redesign the footprint to take both varieties (not that it really works on the mini anyway), the identical pinout means that there should be no routing required, just a footprint swap

tiadobatima, 2021-03-08:
That's very cool @DaneEvans!!
But now I'm definitely lost. LOL 😂 
When looked at the pinout, I could see the pads overlapped a bit (I didn't know it was that well which is cool), but aren't the pins for the mini-e actually different than the mini? From the spec I got I have this:

# Mini
[image]
1. DOUT
2. VSS
3. DIN
4. VDD

# Mini-e
[image]
1. VDD
2. DOUT
3. GND
4. DIN

Since the pins don't match and aren't even symmetric between the mini and mini-e, I'm not really seeing how they could work with the board as is unless these RGBs can work with polarity reversed? What am I missing? 🤔 

Anyways, I ordered some mini-e from Aliexpress a few weeks ago and Plcpcb should be shipping my Sofle board today or tomorrow. Might get it here early next week, and can run a test with the mini-e. Will let you know :)

DaneEvans, 2021-03-08:
Ahh, you're trusting cheap chinese knockoff data sheets - I agree with you on the surface. But there are at least 3 different SK6812 pinouts/pin 1 locations, so my faith in those datasheets is entirely depleted.

~~Do you know which version of the crkbd has the mini-e's, because it's possible that I didn't pull the right one, and I can't find an easy change log.~~

I went the other way, and just compared the footprint to the latest crkbd one, because having worked with these LEDs, it's a lot more trustworthy to use a reference design that you know works, rather than trust the supplier. 

But the crkbd classic, the do-everything symetrical board is apparently no longer the head revision over there, and isn't mini-e compatible, The Cherry version is, and does have a different pinout, (which I think should also be compatible with the rolled pad version of the sk6812, which I have a reel of because of bad suppliers). 

Sorry for the bad info earlier, I hadn't realised that the crkbd directory layout had changed so much since I built my travel board. 

I'll do the new footprint tonight, and talk you through swapping the footprints over, but you'll need to handle the last 5mm or so of routing to each LED.

DaneEvans, 2021-03-09:
Hey @tiadobatima, I've made you a board to start on on my repo (https://github.com/DaneEvans/SofleKeyboard/tree/mini-e) 

I've made the footprint, placed it on the board, locked the switches and stripped back the nearby tracks, so it should be fairly simple for you to route. 

Please check the footprint (SK6812MINI-E_and_cherry(mini-pinout) )  - note that the numbers are different - this is because kicad apparently doesn't support pin tables, so the only way to remap it is to change the numbers on the schematic.  Because I want to keep the schematics the same, I've chosen to mess up the footprint numbering reletive to the mini-e datasheet. The net names match however. 

After that you are free to route the new mini-e version of the pcb, 
Create a mege request whenever you like and I'll do a quick review, clean up any documentation, and make it compatible with what Josef has done before sending it over here.

tiadobatima, 2021-03-10:
Very cool @DaneEvans ! Thank you very much!
I'm gonna take the rerouting process a bit slowly because I already placed an order for the current RGB version, so I'll likely focus on getting that built first because my shoulder is really killing me lately :D
I'll try to put a few minutes of work on those tracks on a regular basis, and loop you in before I go too far in case I'm doing something bad.

I'll likely be building at least a couple more later, and when I do I plan to use the mini-e.

Thank you for all the help!
g.

tiadobatima, 2021-03-16:
Howdy @DaneEvans... Not sure if you have permissions, but would it be possible to push this branch to the `josefadamcik` org?
Apparently Github doesn't let us have two forks of repos in the same "network" (same name) under the same org/account. Normally I'd just kill my fork from `josefadamcik` and fork yours, but I have a PR currently open against `josefadamcik`.

josefadamcik, 2021-03-16:
@tiadobatima he doesn't have permission to merge. This is only an issue, no branches are linked to it. Do you mean the PR #33 ?

josefadamcik, 2021-03-16:
@tiadobatima Oh, I actually misunderstood what you mean, like completely :D My bad, just ignore me.

tiadobatima, 2021-03-16:
Thanks for merging that PR @josefadamcik !
Now I can fork @DaneEvans' repo safely :)

Thanks!

joecabezas, 2021-03-26:
o what's the next step?, currently, Sofle RGB pinout is not compatible yet with `SK6812-MINI-E`?, the way I see it, it is not, but the cutout is compatible since the `mini-e` is smaller

DaneEvans, 2021-04-07:
@joecabezas Someone takes the time to route the branch on my fork, at the moment it sounds like noone is in a rush to do it

- [ ] Pull my fork
- [ ] Route the last 5-10mm of all leds
- [ ] commit 
- [ ] I'll review it 
- [ ] Generate Gerbers and documentation 
- [ ] Merge it onto this fork as the head

joecabezas, 2021-04-09:
thank you, I am working on my own version, kailh choc compatible

DLopezGo90, 2021-05-21:
@DaneEvans bro is SofleKeyboard_mini-E.kicad_pcb ready to be used?

DaneEvans, 2021-05-21:
No

tiadobatima, 2021-05-24:
@DLopezGo90 It's hard to find time, now that I got my Sofle RGC working, I'll try to set some time aside to get the Mini-e version going. I'll update this thread when I get something worth sharing.

DLopezGo90, 2021-05-25:
@tiadobatima Thanks bro, I'll be expectant

brianlow, 2021-06-11:
If it helps, I have another Sofle variant (Choc switches) that uses SK6812-MINI-E here:

https://github.com/josefadamcik/SofleKeyboard/issues/40

They are pretty easy to solder though I haven't tried the non-MINI-E.

tiadobatima, 2021-06-16:
Howdy folks... Sorry for taking this long, but I thought it would be better to also change the underglow and indicator lights to mini-e as well from @DaneEvans amazing initial work on this, so people have to deal with only one type of RGB. I also found that soldering the regular underglow/indicator minis on the SMD was a LOT more challenging than the drop-in per-key RGBs.

Here's the PR #90 . This should be considered a rough draft, since I'm new to Kicad, I haven't printed the PCB yet, and it's been over 20 years since I last worked on a PCB design. I definitely need adult supervision here.
I'm happy to make any change you guys recommend.

utlandr, 2021-08-19:
Just read @josefadamcik's  [cool article](https://josef-adamcik.cz/electronics/another_year_for_sofle.html) and wanted to mention that I am in the process of building the mini-e variant of the RGB Sofle. Its early days, but the PCB has been manufactured, just waiting on other bits and pieces before putting it all together. 

Will keep y'all posted.

page200, 2021-08-21:
@utlandr Thanks, looking forward! Are you using https://github.com/tiadobatima/SofleKeyboard/tree/mini-e ?

utlandr, 2021-08-22:
> @utlandr Thanks, looking forward! Are you using https://github.com/tiadobatima/SofleKeyboard/tree/mini-e ?
@page200 Yes. I am using the one in PR #90

gknapp, 2021-08-24:
I've been enjoying my Sofle RGB build. Last night I started adding indicator and drop lighting LEDs but became a bit unstuck. When I first soldered the indicator LED it lit up green, albeit dimly. I added another SK6812-mini-e LED on pad 4 (per the build guide). That also dimly illuminated green. After adding a third, the LEDs on the bottom side would not illuminate and the indicator LED glowed red. I also noticed the back of the ProMicro was warm (but still shows as a Sofle and hot swap sockets work).

I've been mounting the LEDs with the [notched corner & leg](https://shop.beekeeb.com/wp-content/uploads/2021/02/shop.beekeeb.com_sk6812mini-e-1-1536x1536.jpg) (`GND`) on the outlined pad (`DOut`), assuming it's pin 1 on the PCB per the guide?

Seeing #90 I assume these LEDs aren't compatible with my [v2.1 board](https://customkbd.com/collections/complete-kits/products/sofle-keyboard-kit-v2-1-rgb)? (bought the kit May 5th)

I'll try some SK6812 mini 3535 I have spare instead. Many interesting forks / evolutions have been created recently 👍

josefadamcik, 2021-08-24:
Hi @gknapp ,
why did you decide to add comment to this issue? Have you even read the original post? The only case when it makes sense to comment on existing Github issue is when you are reasonable sure you have the same issue as the OP and you can provide some information regarding the issue. In any other case you should create a new one. This is a general rule of thumb which would apply to any project on Github.

Please, create a new issue with your problem if you feel you need more help. This is going only to confuse everyone. I would kindly ask you to delete your comment as soon as you read my answer, I'll delete my post as well later.

BTW, You are right. The Sofle RGB V2.1 **does not** use SK6812-mini-e. This issue (you are commenting on) is about creating a new variant of Sofle RGB which dose add such support and the discussion above even mentions they are not compatible (they don't have some pinout). 

This feature haven't been released yet (which you already guessed) and because you have indeed bought the original Sofle RGB V2.1 you have to use the components stated in the build guide (SK6812 mini). 

I know there's a lot of those small LEDs and it's hard to navigate the space. My first (and only) attempt to build a keyboard with RGB ended when I learned I have wrong RGB LEDs :D

trougnouf, 2021-11-22:
Can I use SK6812 Mini-E with the Sofle RGB as it is published in the Github repo? Or a mixture of both Mini-E and Mini? I've already ordered the PCB and I am now getting the components.

Sorry if this has already been talked about and I didn't catch the information.

DaneEvans, 2021-11-22:
Please re-read this thread.

pfn, 2021-11-28:
> > @utlandr Thanks, looking forward! Are you using https://github.com/tiadobatima/SofleKeyboard/tree/mini-e ?
> 
> @page200 Yes. I am using the one in PR #90

Any updates on your trials @utlandr ?

utlandr, 2021-11-30:
Hi @pfn! As of about two weeks ago I have **finally** received all the pieces to the puzzle. I unfortunately haven't found the time to put it together since then.

That said, I hope to at least get the LEDs this weekend. I am a complete novice when it comes to soldering and electronics so expect some to see a picture of magic smoke before pretty lights :+1:

pfn, 2021-11-30:
> Hi @pfn! As of about two weeks ago I have **finally** received all the pieces to the puzzle. I unfortunately haven't found the time to put it together since then.
> 
> That said, I hope to at least get the LEDs going this weekend. I am a complete novice when it comes to soldering and electronics so expect to see a picture of magic smoke before pretty lights 👍

That's awesome that you got all your parts together @utlandr best of luck in your build! I am settling for a standard sk6812-mini build for now, and maybe will consider mini-e in the future.

dmikhalsky, 2022-01-29:
Any news on this?

KlausHans, 2022-10-23:
Still no news? I am interested in this build.

taydevr, 2023-02-03:
Hi @utlandr, 

Can you share the results of the tests please?

I am interested in using this version =)

utlandr, 2023-02-06:
@taydevr check out the PR (#90) for the latest successful builds.

taydevr, 2023-02-06:
@utlandr Thanks!!!

closed 2025-06-18 by DaneEvans