josefadamcik/SofleKeyboard #47: JP9/10/11 are not mentioned in the build guide

baruch, 2021-03-13:
The V2 PCB has these jumpers but no mention of them in the build guide. It seems that one should choose if he wants serial or I2C over the TRS/TRRS cable and that defines which cables can be used and I'm left wondering what should be the criteria to make the choice besides the cable (I will have TRRS cable). Will there be a benefit for using I2C over serial in this case?

josefadamcik, 2021-03-13:
@baruch Short answer: there is no benefit and no reason to do any choice. Just go with the default.

Longer answer:

They are not  mentioned mainly because it's not necessary nor advised to do anything with them. It's only part of the design we are inheriting from older designs and they are not usually documented (e.g. corne, lilly, helix - afaik their documentation also didn't mention what to do with those jumpers and when).

My plan was to eventually experiment with i2c and determine which one to use on my own.

But  according to my research among QMK community members, the serial protocol is considered to be better than i2c for QMK split keyboard. There are several reason for this which are discussed over and over in the QMK discord and I bet I don't even remember all of them - just from the top of my head there were problems with reliability with longer cables (coiled etc.) and multiple problems with OLED displays connected at the same time on i2c bus.

If I ever get to version 3, I would probably remove i2c support completely and remove all the oddities around (TRRS cable and connector which is not needed).

baruch, 2021-03-13:
OK. But it looks like I need to close JP11 to get serial on V2.0.1, is that correct?

josefadamcik, 2021-03-13:
@baruch No, that is not correct. As I said above, there's no need to do anything with any of those jumpers. That is why they are not mentioned in the build guide.

If you open the pcb in KiCad, you can see that the jumper is already bridget. You would have to cut the trace between pads if you wanted to switch to i2c.

baruch, 2021-03-13:
OK. Thanks!

closed 2021-03-13 by baruch