josefadamcik/SofleKeyboard #50: Is Sofle v1, compatible with choc v2?

joecabezas, 2021-03-19:
Hello!, thanks for this project, one question:

is Sofle PCB v1 PCB:
https://github.com/josefadamcik/SofleKeyboard/blob/master/Sofle_v1/PCB/SofleKeyboard.kicad_pcb

compatible with choc v2 key switches?
https://www.aliexpress.com/item/4000803757746.html

josefadamcik, 2021-03-19:
I don't know. Sofle V1 uses Kailh sockets for Choc switches which were designed for version one at times when no version 2 existed. You need to check if they have compatible pinout and sizes. Kailh should be able to tell you if they are compatible somewhere in their specs or marketing documents. Or try to google around. 

If you share what you learned with us here, it would be very appreciated!

josefadamcik, 2021-03-19:
@joecabezas Actually when I look at the link again, it seems they have for some reason 3 pins (contacts) at the bottom and require one additional hole. That can't be compatible with version one which has only 2 pins for sure. So yeah, I am quite sure the answer is no.

joecabezas, 2021-03-19:
yeah, also the middle plastic is way bigger in choc v2, thanks again for taking a look, you are awesome

closed 2021-03-19 by joecabezas