josefadamcik/SofleKeyboard #51: Can't load footprints

joecabezas, 2021-03-19:
how can I fix this?, it seems like there are absolute references to a path only the author have, how can we fix this?, or can those footprint be included in the repo?, thanks

[image]

josefadamcik, 2021-03-19:
All the footprints should be in the repository under footprints. Even the Keebio-Parts are included (as a git submodule, I believe). So just make sure they are properly checked out and change paths in the KiCad footprint library.

To be honest, you might not even need it unless you change something. But I am far from being a KiCad guru.

joecabezas, 2021-03-19:
thanks, I just noticed that, thank you, I am closing this

closed 2021-03-19 by joecabezas