josefadamcik/SofleKeyboard #54: Are capacitors missing on the SK6812MINI?

joecabezas, 2021-03-21:
I am reading the datasheet of the SK6812MINI, and in the `The typical application circuit` section it suggests to put a capacitor on each LED:
https://cdn-shop.adafruit.com/product-files/2686/SK6812MINI_REV.01-1-2.pdf

it seems like is possible to avoid it, but others suggests is better to have one:
https://electronics.stackexchange.com/questions/392925/can-i-replace-these-many-capacitors-with-just-one

Is there any specific reason the RGB version does not include capacitors?

DaneEvans, 2021-03-21:
In short, probably laziness.

No other keyboard I've seen or built has them, most likely because the
pioneers didn't want to solder another 60ish passives.

It's probably a good idea, although a PC power supply probably has a lot of
capacitance itself.

  I don't know about one on every LED, but some bulk capacitance on board
would reduce microcontroller hangups when you plug the USB in. I've never
had the micro brownout when switching all the LEDs to White, or had them
fail to read the Din signal which are usually the symptoms that individual
caps are designed to correct.

It's a 'won't fix' at this point.  It's a proven design, and the same as a
lot of other keyboards. If I do a new spin of the board I may add
footprints for them, but it won't go into the recommended build

joecabezas, 2021-03-22:
thank you for the explanation <3

also, the capacitor is not intended to protect the the (pro)micro, but the controller inside the LED, which can cause on animations to flicker, still thanks for answering and confirming it is laziness, I wanted to also be lazy on my design, please have a good night, I am closing this

closed 2021-03-22 by joecabezas