josefadamcik/SofleKeyboard #56: SofleRGB various questions

dudeyds, 2021-04-02:
Apologies as I'm not so sure this is a bug with anyone's code/ schematics, please remove if it's inappropriate, I wasn't sure where else to go for help.

Recently I completed my SofleRGB build, My first split keyboard so I'm very excited!

I'm having some issues with the RGB LEDs on the right hand side, I wonder if anyone had this issue  before?

https://youtu.be/8aOsMrpwBoo

As you can see the left hand side is working perfectly fine (I don't have Underglow LEDs, just the indicator and per key backlighting)

I couldn't find the defaultRGB layout in QMK, perhaps it wasn't pushed? So I looked through and found Dane Evans' rgb_default in his upstream branch of QMK. That's the one I'm using here.

I changed the RGBLED_NUM to 60 and the RGBLED_SPLIT to {30 , 30}. Other than that everything else is the same.

So far I've tried swapping the pro micros and flashing them, same problem, re-soldering the LEDS, no luck there.

The lights on the left are static if there's no animation, though strangely if you press a layer change button it randomly changes all the lights. Perhaps I have broken something in the layout file?

And just to confirm all the keys work as expected, the only issue I'm having is the LED's

DaneEvans, 2021-04-02:
Your right hand side has a soldering bug. Probably in the first LED that's
flickering.
It could also be a power supply issue, so check it plugged in by USB
first.

The left hand one is behaving properly.  The LEDs are set up as a layer
indicator, but the led locations are set up for a board with undergow. Find
the macros at the top of keymap (I think), and fix them up.

I pushed it, it's still waiting on a merge request to be actioned I think.

If it hasn't been done I might add the dynamic numbering to those macros
that I had planned, so it's more adaptable to your setup then resubmit, but
it won't be for a while

dudeyds, 2021-04-07:
Thanks for the feedback, the issue was indeed the first LED, I've gotten everything working now, now to start messing with the layout! thanks so much for your help

closed 2021-04-07 by dudeyds