josefadamcik/SofleKeyboard #57: SofleRGB - Which pins to short?

sdkrocks, 2021-04-05:


DaneEvans, 2021-04-05:
Nothing should blow LEDs, apart from the seperate LED header (J1), all of the pins on the LED select 2.54mm headers have only signals on them. 

Pin 1 has the square pad on the headers.  But I have a feeling that you are having issues with your led orientation rather than the headers.

sdkrocks, 2021-04-06:
> Nothing should blow LEDs, apart from the separate LED header (J1), all of the pins on the LED select 2.54mm headers have only signals on them.
> 
> Pin 1 has the square pad on the headers. But I have a feeling that you are having issues with your led orientation rather than the headers.

Well thanks for the amazingly fast reply, and clarification of the pins. I double checked and it looks like I was putting on them backwards!

closed 2021-04-06 by sdkrocks

DaneEvans, 2021-04-06:
No worries, hopefully it's working for you now.

sdkrocks, 2021-04-06:
Well unfortunately after spending multiple hours on it, I cannot seem to get them going.
By default - in your repository the indicator light is turned on correct? I've tried soldering in multiple ways - but no signs of life. Even using the lowest temp I can. 

They are not blowing anymore - so at least that is a good sign.

DaneEvans, 2021-04-06:
They should be, if you have a length of RGB strip you can test it in the
RGB header. Which confirms whether soldering is the issue.  I usually check
this first to verify software.

If you upload some pictures I may be able to help, but at the moment it's
all guesswork.

If you have drop or per key lights installed, I'd try bypassing the
indicator, because the signal always stops at the first bad LED.

I'll say again, the most likely situation I can see is that you have the
wrong LED orientation, which LED do you have? Pictures, there are at least
4 different designs and patterns that all call themselves minis

sdkrocks, 2021-04-07:
Here is the direct link to the ones I'm using - https://www.aliexpress.com/item/33019583218.html
I have the black mini ones.

The bottom pads match the ones in the picture on the guide. 
Here is the pictures including how I attached the LED and it flipped over to the left.
https://imgur.com/a/8BgNZ0f

I promise the first couple times before this one it looked much better - the leftover flux and the low temperature made it look bad. Looking under a microscope they all seem to be making contact and attached.

I have the v2.1 board, and the pictures are on the left half. I will try the RGB header later when I can.

Thanks so much for all of your help.

DaneEvans, 2021-04-07:
You're bypassing the indicator LED, use tweezers to short the solder bridge on J6 to the unsoldered pin and the LED should come to life

sdkrocks, 2021-04-07:
Wow. Well you are a life saver, it immediately came on. 
I was following the table in the guide, where it says short pins 1-2 on each if you want all of them enabled. 

I'm trying to get all of the LEDs enabled - (Indicator, per key, and drop)

So for all of them to work just short the ones with the silk screen, correct?

Thanks so much! I appreciate your replies.

Thanks so much! I really appreciate o

DaneEvans, 2021-04-07:
Glad it's working!

Yeah, looks like that was the only place I had it wrong, the silk screen,
the schematics, the build guide diagram, and text all had 2-3,

It was just the one column in the table. There's a change request in to fix
it up.

You didn't think that it was odd that you were bridging where the
silkscreen said BYP, not where it said IND?

The silk screen should always be correct (now, less so on the v1.0 that I
have), the problem is just in translating that to what they do, and the
order in which they operate, which is why the diagram is in the build guide

sdkrocks, 2021-04-07:
> Glad it's working! Yeah, looks like that was the only place I had it wrong, the silk screen, the schematics, the build guide diagram, and text all had 2-3, It was just the one column in the table. [...]

Haha yeah I guess not, that was what I originally was going to do but once I checked the guide I changed my mind.

Thanks a ton once again.