josefadamcik/SofleKeyboard #58: Issue Template

DaneEvans, 2021-04-06:
To make it easier to get any issues answered, or to triage  them as a dev/user issue we need more info than many people are giving us, to make this easier, adding a template to the new issue page would be useful - note, I don't actually know how to do that, just that it is possible, and that I won't have the rights to do it here. 

Proposed:

**Which Board?**
_Original / RGB_  

**Which Board Revision?**
_This is the one written on main pcb that you are using. 
On the original it is written on the top outside corner on the silk screen. 
On the RGB it is written on the bottom outside corner, in copper under the silkscreen,_
Original = 1.0, 2.0 
RGB =  2.0, 2.1

**What is the issue?**
_Please also include what other components you are having issues with - is the software QMK or the configurator, include pictures of your LEDs or soldering issues etc._

**What you expected:**

* [ ] item 1
* [ ] item 2

DaneEvans, 2021-04-07:
```
**Kind of issue?**
_Bug / Assembly / Software / HELP!_

**Which Board?**
_Original / RGB_  

**Which Board Revision?**
_This is the one written on main pcb that you are using. 
On the original it is written on the top outside corner on the silk screen. 
On the RGB it is written on the bottom outside corner, in copper under the silkscreen,_
Original = 1.0, 2.0 
RGB =  2.0, 2.1

**What is the issue?**
_Please also include what other components you are having issues with - is the software QMK or the configurator, include pictures of your LEDs or soldering issues etc._

**What you expected:**

* [ ] item 1
* [ ] item 2

```

josefadamcik, 2021-04-07:
I have created the first one in #64. In the end the settings UI just creates a commit. 
We could also have multiple of them - maybe we could  have 4 of them for the 4 kinds so the user picks this one by picking the right template. 

I think you can just create pull requests with changes to the file and/or add more if you wish.

closed 2021-04-07 by DaneEvans