josefadamcik/SofleKeyboard #62: Encoders skip when plugged into the off hand

DaneEvans, 2021-04-07:
from #19 

> If you are going to be using encoders, expect to be plugging the keyboard only into the part that has your encoder. Encoder from the other half (the one that is not connected with a usb cable cable) will be skipping steps and generally be unusable for any smooth operation. There's a QMK bug report somewhere, but I can't find it now.

DaneEvans, 2021-04-07:
this was fixed in https://github.com/qmk/qmk_firmware/issues/7055. 

Whether we are using that code or duplicating it I don't know, We should do a cull to get as close to pure qmk head as possible

josefadamcik, 2021-04-07:
@DaneEvans Sofle is using core QMK for all split related things encoders included. I threw away all the copy pasted legacy which was present at the firmware for Lilly58 at the time I "forked" the board. So it should be fixed for Sofle as well.

To be honest, I wanted to invest some time into testing both claims in #19 but never got to do it (I forgot about them eventually).

DaneEvans, 2021-04-07:
Then it can be closed, I knew the crkbd and lily had old stuff, and I never looked at yours to see if it was all removed, so I figured I'd leave it for you tell me

closed 2021-04-07 by DaneEvans

ryuugaplays, 2022-05-25:
I know this is an old issue and seems to have been solved, but Ive been playing with the encoder recently and Id like to report that I have been experiencing this exact issue with the slave side feeling less responsive than the master side. I tested this by putting both sides as mouse cursor then while turning each side individually you can see the mouse is ever  so slightly unresponsive when turning the slave side.

This can be further exacerbated and more obvious if you try this in a video game. load up any FPS and  try to turn the camera with either side youll find that the slave side just doesnt want to move properly.

brianlow, 2022-05-25:
Thanks for reporting 

Some ideas
- Connect right half to computer. Does the problem move to the left half (now the slave)?
- Compile version of firmware with minimal features and flash both sides. Maybe a feature like animation consumes too much bandwidth. 
- Make sure on latest version of QMK
- Could look at I2C communication between halves. I understand that is faster. Would need hardware changes though

ryuugaplays, 2022-05-29:
Hey brian thanks for the suggestions, sorry it took a few days but I finally done the tests and here are the results:

> - Connect right half to computer. Does the problem move to the left half (now the slave)?

yes problem persists regardless of handedness

> - Compile version of firmware with minimal features and flash both sides. Maybe a feature like animation consumes too much bandwidth.

I flashed rgb_default but made sure to turn off RGB lights {RGBLIGHT_ENABLE = no} and OLED{OLED_ENABLE = no} in rules.mk problem still persists. Not sure if there are any other features I should disable, lemme know if there are. firmware size was 18212/28672 (63%, 10460 bytes free) after disabling 

> - Make sure on latest version of QMK

I made sure to update BEFORE doing any of the above, problem still persists

> - Could look at I2C communication between halves. I understand that is faster. Would need hardware changes though

sorry unable to test this as I am unsure about what you mean here

Let me know if there are any other things I can try. I can even post a video showcasing the issue if that is more helpful

brianlow, 2022-05-31:
Can you open a new issue? I apologize I should have suggested that originally. This one is closed and the particular issue resolved.

Not sure I can help either way though. 

* QMK Docs on serial and I2C: https://github.com/qmk/qmk_firmware/blob/master/docs/feature_split_keyboard.md
* Came across his that suggest serial is faster than I2C: https://www.reddit.com/r/ErgoMechKeyboards/comments/qsfega/step_by_step_guide_to_convert_sofle_v2_from/
* If you want to try I2C, will need resistors for R1, R2 and be sure to have a TRRS (no TRS) cable
* Could try faster serial: `#define SELECT_SOFT_SERIAL_SPEED 0` as mentioned in the QMK docs