josefadamcik/SofleKeyboard #70: Documentation issue (RGB)

sdkrocks, 2021-04-08:
**Kind of issue?**
Documentation

**Which Board?**
RGB

**Which Board Revision?**
RGB =  2.1

**What is the issue?**
In the build guide where it is displaying the RGB connections to the PCB, the table is broken and just a single line of text

I also believe that the other table and the diagram and table might be wrong indicating which pins to short. 
The diagram states short pins 1-2 to enable drop lighting, while the silk screen states short 2-3 for UND&BL

Essentially for J4 to instructions are reversed

**What you expected:**

Documentation to be accurate when compared to the PCB

sdkrocks, 2021-04-08:
#71 If I am correct about the pins being wrong on the diagram and table should fix these issues

DaneEvans, 2021-04-09:
closed as I merged the PR, and then fixed it up to actually match the schematics

closed 2021-04-09 by DaneEvans