josefadamcik/SofleKeyboard #75: How to troubleshoot LED issues on SofleRGB

mvk15, 2021-04-24:
**Kind of issue?**
HELP!

**Which Board?**
RGB

**Which Board Revision?**
RGB = 2.1

**What is the issue?**
Have some issues with Drop Lighting:
1. Right half - the 7th LED is flickering.
Tried to reflowed all pins of the 7th LED and 6th LED - didn't help.
Replaced the 7th LED to a new one. Both points haven't helped - the 7th LED is still flickering.
2. Left half - don't work LEDs 5, 6 and 7. Don't know from where should I start to check and what to do.
Tried to figure out the board schematic in KiCad but unfortunately without any success.

If possible please help with fixing and/or explaining how to troubleshoot such issues.

DaneEvans, 2021-04-24:
On the right hand one, replace 6.  It's always either the last one that
works, or the first to fail.

On the left hand side, the same thing, but there's also a disconnected via
(I think that's that side and around those numbers.) So make sure you read
that and see if it applies.

Also check that the LED orientations are all correct.  But usually you just
replace the first to fail, and then the last working one.

 If you want to check you can also bridge across to a data line with a wire
from the data pin of the LED header

mvk15, 2021-04-25:
Thank you for helping.

About the 2nd issue with the left half - used your suggestion. Took a wire and shorted 5th LED's pad DIN with data pin of LED header and found that it doesn't work. However, the 6th and 7th LEDs have started to work when I touched the 6th LED's DIN pad. So the issue was with the 5th LED - reflowed its pads and it started to work.

About the 1st issue - replaced the 6th LED but the 7th LED of the right half is still changing its colors. And moreover, after fixed the 2nd issue related to the left half, its 7th LED also started to change its color.
Could it be that is a false alarm and it is by design? I mean when I enable both halves of the keyboard then 5 drop lighting LEDs on each half have red color but the 6th LEDs are changed their color synchronously.

DaneEvans, 2021-05-07:
You haven't set the numbers of LEDs correctly in keymap.c

By default (ie. mine) the drop lighting is on (single colour, layer indicator) and the per key is on (set to the patterns, I think it defaults to a rolling hue value)

if you enable the indicator led in code, then everything should make more sense

mvk15, 2021-05-07:
Thank you for your help.
I have already touched `keymap.c` file but haven't figured out with RGB part.
Now you have given me a direction and I will start to dig.

closed 2021-05-07 by mvk15

DaneEvans, 2021-05-07:
Look in the macros at the top.
Indicators should be one led starting at 0,
Undergow 6, starting at 1, etc.

Layer ID will also need to be updated, or replaced depending what you want
to do with your LEDs