josefadamcik/SofleKeyboard #82: LED 9 isn't working (drop lighting + per key lighting)

DLopezGo90, 2021-05-25:
LED 8 is working but not LED 9. I don't know what's happening

[image]

For your information I made this bridge (J6) because I don't want Indicator LED

[image]

Help please

DaneEvans, 2021-05-25:
Replace 8.
If you cook them, sometimes it's the output that dies.

DLopezGo90, 2021-05-26:
It didn't work. I replaced LED 8 and 9 and nothing happened

My solder skill isn't bad. I have soldered many Corne Keyboards

You sure LED 9 comes after LED 8? You sure sure they are connected?  DOUT pin goes with silkscreen mark right? Because that's what I did

DOUT is elbow shaped:

[image]

You can see I soldered the elbow shaped contact of the LED to the silkscreen pad on the PCB:

[image]

[image]

I have the version with te unconnected via for NET34 and I had to the bridge on the front side to make LEDs 6 and 7 work:

[image]

DaneEvans, 2021-05-26:
Then I'd make sure that you have the later LEDs enabled and doing
something.

There are no other known board issues. Plenty of user ones though

DLopezGo90, 2021-05-26:
Should I solder later LEDs? will LED 10 work if 9 doesn't?

DLopezGo90, 2021-05-26:
Bro I found out what was the issue 

This is the config.h file of sofle keyboard via keymap:

[image]

It will only make 14 LEDs work (7 on each half). It needs to be modified to 72 I think (36 on each half):

[image]

This should be added to the build guide?

DaneEvans, 2021-05-26:
The default one has them all enabled. I don't know where you pulled that
file from, but it's not one of the ones I've released.

DLopezGo90, 2021-05-26:
Bro where I find the ones you have released?

DaneEvans, 2021-05-27:
Eventually the QMK repo, once I make it through their review process,  but
for the moment, the master or upstream branches, or any tagged commit on my
fork of qmk.

t321, 2021-06-10:
As you said @DLopezGo90, only the led until 9 works if you use the via hex, in the readme says that only underglow is enabled (https://github.com/qmk/qmk_firmware/tree/master/keyboards/sofle/keymaps/via).. I also flashed VIA and had the same issue. 

If you install qmk and compile with `qmk compile -kb sofle -km rgb_default` and flash this hex, you will have all the led configured. Don't worry if the leds from the external colums and rows are red, are configured in this way :P
[image]

If you want to avoid the compile steps.. you can use the hex that I created (I had to compress in zip because GitHub doesn't allow `*.hex` files): [sofle_rev1_rgb_default.zip](https://github.com/josefadamcik/SofleKeyboard/files/6633459/sofle_rev1_rgb_default.zip)

fi0, 2021-06-16:
Is there a via firmware that enables all LEDs?

DaneEvans, 2021-06-16:
No idea.  The only one I have provided is the qmk repo.

Any you compile yourself from the qmk repo from rgb_default, or my keymaps
should have all or most LEDs on by default.

fi0, 2021-06-23:
@t321 How do I set the outer column to the same colors for the rainbow mood?

DaneEvans, 2021-06-23:
Remove the outer column from the SET_LAYER_ID macro.

In remove lines 60 and 61

fi0, 2021-06-25:
@DaneEvans I compiled the rgb_default keymap and found that the last LED didn't light up on both left and right side. 
I've soldered the indicator, per key and underglow LEDs.

closed 2024-03-12 by DaneEvans