josefadamcik/SofleKeyboard #83: Pro micro don't work when is connected to the PCB

danieltoro, 2021-05-26:
Kind of issue? Assembly 

Which Board? RGB

Which Board Revision?  RGB = 2.1

What is the issue? Pro micro won't work when I conected to the pcb, either i can't flash with the pcb reset button

[image]
[image]

I don't know what to do

DaneEvans, 2021-05-26:
Do the LEDs stay lit?
If yes, it's a reset issue.
If no, it's a power issue.

Is the pro micro on the correct set of headers for the side of the board?

On the right half, micro on top of board, USB down = right set of headers

On the left half, micro on top, and USB down = right set of headers.

I'd need to look at the schematics to work out other configs.

danieltoro, 2021-05-26:
this is the pcb for my left hand 

[image]

DaneEvans, 2021-05-26:
Assuming you are putting the micro on with the components towards this PCB,
it's right.

Is the micro losing power (no lights showing)?

Did you test it before putting the LEDs on? If they are around the wrong
way it can cause the micro to lose power. That's probably the most likely
from what I've seen before.

Liilnopitynope, 2021-05-27:
I am having the same issue where pro micro alone is working but once I plugged it in the PCB it's not working. The LED indicator on the pro micro itself is lighting up, but my pc is not recognizing the board, but none of the other LEDs on the board is working.

danieltoro, 2021-05-31:
> I am having the same issue where pro micro alone is working but once I plugged it in the PCB it's not working. The LED indicator on the pro micro itself is lighting up, but my pc is not recognizing the board, but none of the other LEDs on the board is working.

danieltoro, 2021-05-31:
@Liilnopitynope can do you make it work?

DLopezGo90, 2021-06-02:
My right half is working but left isn't 

Right half (working):

[image]

Left half (not working, Pro Micro doesn't turn on and gets hot):

[image]

Should I put the Pro Micro with the components facing upward (not toward the PCB) on the left half?

[image]

Help please

DaneEvans, 2021-06-02:
It's getting hot because you have a short.  

2 options I can see, wrong header, or an LED short

Which set of headers are you in on each half? 

If you want both micros in the same orientation, then you need one to be in the outboard set, and the other in the inboard.  

Did you test before adding LEDs? 
Did you test with only a few LEDs? 

If you continue to refuse to try those two things, I suggest you remove the 5v and LED pins grow your micro, that should allow you to test the key matrix and confirm that it's an LED short.

DLopezGo90, 2021-06-02:
I have both Pro Micro with the components facing the PCB and they are in the inboard set (with inboard you mean the holes inside the white silkscreen rectangles right?)

C'mon bro, I posted images. You can tell me by looking at them

I followed the guide, it says that both Pro Micro must have the components facing the PCB and that holes inside narrow white rectangles must be used to put the Pro Micro

[image]

Help please

DLopezGo90, 2021-06-03:
[image]

I have both Pro Micro with the components facing the PCB. I did it wrong?

DaneEvans, 2021-06-03:
Ok. I'm done here.
You upload crappy images with  angles that don't even show the placements,
and want me to debug this for you from a timezone across the world.

While ignoring all of the questions that would lead us to a solution and
giving me attitude.

The only reason I have left it open this long is that I don't yet have
confirmation that anyone has a fully working build of this version of the
build.

Given that one of you has a working left board, and the other the right, I
think we can rule out a PCB issue, which means you're on your own.

DLopezGo90, 2021-06-03:
I looked at danieltoro's images and it's clear that he used SK6812 Mini-E which has a different pin distribution and is probably the reason why he's having issues. I thought you were ignoring images because you didn't notice but if that's not the case I apologize

I used this version: https://github.com/josefadamcik/SofleKeyboard/blob/master/Sofle_RGB/PCB/SofleKeyboard.kicad_pcb 

[image]

I can see it's the 2.0 version and it has that unconnected via for NET34 issue

[image]

So I had to do the bridges:

[image]

[image]

In the beginning both halves were working but suddenly left half turned off and didn't work anymore

[image]

I don't  think it's a PCB issue because there's a guy in reddit who uses this Sofle RGB keyboard without problem

closed 2021-06-08 by danieltoro