josefadamcik/SofleKeyboard #86: Question: are 2.0.1 version and RGB version merged now?

Geobert, 2021-06-04:
Hi, not a help request :)

I was wondering if the 2.0.1 version in the release section supports RGB.

Also, it is possible to install LED later? As it's my firt build, I think I'll aim to have a working keyboard first, and avoid LED to reduce the amount of soldering, but my GF fancy some lights on her one so I'm trying to see how I will do it, and when :)

josefadamcik, 2021-06-04:
> I was wondering if the 2.0.1 version in the release section supports RGB. 
No.

> If not, where can I find RGB gerber files? 
https://github.com/josefadamcik/SofleKeyboard/tree/master/Gerbers/RGB

> Also, it is possible to install LED later? 

I don't know exactly since I haven't' build the RGB version myself, but I assume it should be possible. Just check the PCB for places where you might loose access to the LED after soldering something else. But  since the switches are socketed, it' should be ok.

Geobert, 2021-06-04:
Thank you for your insight!

closed 2021-06-04 by Geobert

DaneEvans, 2021-06-04:
You can definitely install them later, or not at all. I usually do to check
that everything is working before i do the LEDs.

there are also headers if you only want the drop lighting, a strip of leds
is trivial to add.

Geobert, 2021-06-04:
> 
> You can definitely install them later, or not at all. I usually do to check that everything is working before i do the LEDs. there are also headers if you only want the drop lighting, a strip of leds is trivial to add.
> […](#)

Headers? QMK C header file?

I'll need them all but the indicator one. My GF want her keyb to shine x)

DaneEvans, 2021-06-04:
No, a 3x1 2.54mm pin header. It saves all the soldering if are lazy.

Sounds like you'll just be using the LED footprints then.

Have fun!