josefadamcik/SofleKeyboard #87: [Question] HASL finish on top and bottom

Geobert, 2021-06-08:
Sorry to bother you again with some newbie question :)

I'm considering choosing ENIG finish for the PCB and HASL for top and bottom plates. My question is: as the top and bottom doesn't have any circuit, will they be lead free even if I chose HASL?

Thank you!

josefadamcik, 2021-06-08:
Hi,
there's a little bit of exposed copper around the mounting holes and over
the logo (bottom only06). Those are the only places that will be covered by
HASL. So technically no, it's not going to be lead free. But the covered
surface is very minimal.

Cheers,
Josef

Geobert, 2021-06-08:
Thank you, I'll choose lead free HASL for them then, because I eat at my desk, so I'm concerned about this :)

closed 2021-06-08 by Geobert