josefadamcik/SofleKeyboard #88: Help: Can't get RGB working

lukeraphael, 2021-06-15:
**Kind of issue?**
Assembly

**Which Board?**
RGB

**Which Board Revision?**
2.1

**What is the issue?**
Unable to get the rgb working 

I have checked my soldering and led orientation several times but am still unable to get the rgb working. May I know what I am doing wrong? Below are the photos of my board. Thank you so much.

[image]
[image]

These are the leds im using: https://www.aliexpress.com/item/32885017817.html?spm=a2g0s.9042311.0.0.1b434c4dUy0i5k

DaneEvans, 2021-06-16:
A) are any of them working 
B) have you bridged the indicator bypass jumper (these are designed to be done with a little bare copper or a solder ball, rather than the large wires. It won't cause any issue, but it's a lot neater)

lukeraphael, 2021-06-16:
Thanks so much for the quick reply!

A) None of them are working

B) Yep, I agree it's a lot neater, but wanted to test out the leds before tidying up. I have actually bridged j6 with a solder bridge and the others with wires. I couldnt quite get the solder bridges on the other jumpers hence the wires. I'll clean them up once I manage to get some leds working.

DaneEvans, 2021-06-16:
Is the micro staying powered? Is it programmed? It it outputting to the
LEDs? (This is where a short strip into the LED header is useful)

Are they getting 5v?
Check on any LED footprint or the LED header.

lukeraphael, 2021-06-16:
The pro micro is powered and programmed with the rgb default in the repository. 

1) Attaching a short LED strip did indeed work with no other changes to the board as shown below

2) I don't have a potential meter hence I can't test whether they are getting 5v. Is there any other way to check?

[image]
[image]

lukeraphael, 2021-06-20:
I have found out the issue - An LED is soldered to the indicator position on the underside which causes all the lights to fail. Thank you for all the help!

closed 2021-06-20 by lukeraphael

DaneEvans, 2021-06-20:
Unless it's the wrong orientation, it shouldn't matter which side of the
board it is on.

Glad you found it anyway

lukeraphael, 2021-06-21:
I actually replicated the issue several times as I had multiple boards soldered in the same way. 

Maybe it's worth checking out. I'll try to replicate it again when I build another one.