josefadamcik/SofleKeyboard #94: Help: summarise your issue here.

sdkrocks, 2021-06-25:
**Kind of issue?**
Assembly / Software

**Which Board?**
RGB

**Which Board Revision?**
RGB =  2.1 

**What is the issue?**
The left half of the keyboard LEDs will not come on, but only if plugged into that side.
If the USB c cable is plugged in on the right half, everything works perfectly.

This is my 4th build of the Sofle RGB, and the first one with this issue. All of them run the same firmware. 
What could I do to troubleshoot this? 

I have tried resoldering the first & second LED on the non-working half several times with no avail. Don't forget these LEDs do work if you plug the cable into the right half. 

I have also tried skipping the indicator LED by jumping the other part, but it still does not work. If I swap the Elite C's to opposite sides the issue is the same. I reflowed all the connection points on the board for the Elite C.

When plugging in the cable to the broken left half, the right half LEDs will flash on for a second, then turn back off. All the keys etc. work, just not the LEDs. 

Thanks in advance.

DaneEvans, 2021-06-25:
~~my guess is that you have screwed up the lighting somehow on the master in firmware(unlikely) , you have turned off the LEDS on that half (try the LED toggle key command, if they aren't both plugged in the two halves can get out of whack, or do an EEPROM reset)
 or that the voltage regulator on the left half is dead, so it can take power from the other half, but can't provide the high power needed. 
~~ 

Just read that you tried swapping the elite C's around and got the same results - that's very strange 

I'd try swapping in a known good elite-c from one of your other boards, and/or resetting the EEPROM 

the other option might be a stuck key that has it toggling/turning off the LED's  - although by default that should need 3 keys, which seems unlikely

sdkrocks, 2021-06-25:
Thanks for the reply!

To make sure I had it clear:

If only the left half is plugged in - All LEDs off.
If left & right plugged together and USB c in left, All LEDs off.
If left & right plugged together and USB c in right, All LEDs work fine.
If the Elite Cs are swapped, nothing changes.

Tomorrow I will try 2 different Elite Cs from the working boards, like you said. I've already reflashed them both before.
Key testers show nothing toggled or stuck on. 

I'll also test the voltages and see if anything isn't working in that case.

It's getting late here, I'll report back later :)

kissetfall, 2021-07-09:
Did you try to add `#define RGBLIGHT_SPLIT` into your config.h?

brianlow, 2021-07-31:
Power supply issue? 
* try low value for `#define RGB_MATRIX_MAXIMUM_BRIGHTNESS 75`
* try different TRRS cable

sdkrocks, 2021-08-23:
Sorry I king of forgot about this. 
I ended up fixing the issue by just purchasing new Elite Cs. I guess something was wrong with them? Not really sure to be honest but once they were switched out this was fixed.

closed 2021-08-23 by sdkrocks