josefadamcik/SofleKeyboard #95: Help: LED made a "ploc" sound and died

Geobert, 2021-07-06:
**Kind of issue?**
 Assembly 

**Which Board?**
 RGB_  

**Which Board Revision?**
RGB =  2.1

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
Hi,

It's my first build ever and I tried to solder a LED on Sofle RGB.

I'm without indicator, with backlight and per key.

I've shorted 1-2 on indicator bypass (1 being square header), 2-3 on light select and 1-2 on backlight.

I've soldered the number two, and wanted to test it. I've seen very briefly a red light and heard the LED making a "ploc" sound and died. I can see that the LED "exploded" the black square being not so squared anymore. It took me forever to unsolder it so before trying again, I prefer to ask what happened here? :-/

DaneEvans, 2021-07-07:
Probably around the wrong way, maybe too high a supply voltage (you are
using the Arduino to power it right? ) .
Maybe overheated when soldering.

Taking it off should be a few second process.  Put the front of the led on
a pencil or dowel held vertically, put a blob of solder on your iron, then
roll it around each of the pads.  It's dead anyway, there's no need to be
gentle.

Geobert, 2021-07-07:
Thank you for your answer, as ever :)
For the wrong way, I have triple checked, and pin 1 of LED is marked with the corner being white (the LED is black). And I put the white angle on the pin with a "square" around it. But I'll recheck this evening.

For the power, I think so, I don't have anything on the board apart from the pro micro (with a firmware in it).

I'll try your technique to remove it if needed again, but I'm not sure I'll be able to "roll" the ball of solder, I've already had a hard time to create the short with solder on j4 j5 and j6  ^^'

Also, for the per key LED, how do you guys maintain them at the correct "level" to solder them? masking tape?

DaneEvans, 2021-07-07:
You've read the notes that those LEDs come in multiple varieties, with
different pin 1 locations? There are at least 2 types with the flat back
and irregular pad shapes that are 180 degrees different.

There are also varieties that are rotated 90 degrees, the mini-e and at
least one of the types with rolled pins on the back fall into that
category.

If you plugged it in by USB it won't be a over voltage problem.

Tape works, or I just do them first, so the flat table under them holds
them in place. Or put a book or something underneath if the micro is
already soldered on

Geobert, 2021-07-07:
Yes I read the note and after looking a lot the LED I got look like:
[image]

same pad shape
I also have the sockets already on for the part I'm trying to get the LED on, so it will be tape  ^^' I'll won't do the same mistake on the other part

Thank you again for your insight!

DaneEvans, 2021-07-07:
If they are like that, then you need to put the pin 1 marking on the led at
pin 3 on the board.
Don't blame me, blame the counterfeit manufacturers

Geobert, 2021-07-07:
Oh, that could explain the exploding LED then

DaneEvans, 2021-07-07:
Yep.

Geobert, 2021-07-08:
Let's close this, I'll reopen if I have issue again once I have time to test it :)

closed 2021-07-08 by Geobert

Geobert, 2021-07-11:
I gave up on backlight, too hard to solder for my skill when the switch socket is already there, see my new issue for issue with per key LED :)