josefadamcik/SofleKeyboard #102: 4th row not working on right side

Spaubleit, 2021-07-26:
**Kind of issue?**
Assembly, HELP!

**Which Board?**
Original

**Which Board Revision?**
2.0 

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 

**What is the issue?**
English not my native language and this board was my first serious soldering experience. I'm sorry for silly mistakes.
When I assemble board I noticed that right display not working. This is not a problem for me. But after some test I found that 4th row on my right side not working either. I checked all лов display contacts and found short circuit between SCL and SDA contacts. After removing display the short circuit is preserved on the contact pads of the board. Could it be related to the broken key row? What tests need to be done to determine the problem? Thank you.

DaneEvans, 2021-07-26:
Most likely causes to take a whole row out are a bad row diode, or bad
soldering at the micro.

Check the schematics to figure out which ones to look at.

Spaubleit, 2021-07-27:
Hi. I checked all the diodes and they are fine. But there is no contact between row 4 on controller and switches. And there is issues with display contacts. Is my board broken?

DaneEvans, 2021-07-27:
I doubt it.

The display intentionally needs solder bridges, and there is nothing on the
board that even the cheapest manufacturer should screw up.

 Get the PCB layout and buzz it out if you're worried.  But it will either
be the row 4 diode, which could be either dead or badly soldered, or the
row 4 pin.

Spaubleit, 2021-07-28:
It's not the manufacturer. It's me who desolder broken sockets from the board using air gun. As I said it's my first soldering experience so I made many mistakes. Anyway it looks like I need to order new boards. Thank you for help and this beautiful keyboard. My left side looks fantastic.

closed 2021-07-28 by Spaubleit

DaneEvans, 2021-07-28:
What weight copper did you use?

These should be pretty tough, assuming 4-6 Oz copper.  For hand soldering,
especially with an entry level iron, you want at least that heavy

Spaubleit, 2021-07-29:
I don't really know. Because I ordered complete keyboard set from a store. I was just stupid. I used heat gun set on 250C. It melted my controller sockets and created bubbles under solder mask. Some tracks was ripped off. I was surprised that keyboard is still works. I save my board by the way. Now I have hand wired row :)