josefadamcik/SofleKeyboard #103: Help: How to diagnose this LED malfunction

Geobert, 2021-08-01:
**Kind of issue?**
Assembly

**Which Board?**
RGB

**Which Board Revision?**
RGB =  2.1

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
Hi, I've flashed the main side with `RGBLIGHT_EFFECT_RGB_TEST` to check all the LED and I'm facing this situation, using the build guide numerotation

https://josefadamcik.github.io/SofleKeyboard/images/build_guide_rgb/board-perkey.png

- LED 2 (first in the strip) has no red (I've replaced it to be sure, still no red)
- 3-9 are static purple
- 10 is static blue
- 11 and 12 are static purple
- 13-26 are working ok
- 27 28 are  static purple
- 29 30 are ok

I'm puzzled by the 1st in the strip missing a color, then a bunch of them static, and then some working perfectly fine etc.

Also, sometime, the part that is working fine doesn't light up immediatly but only after 2 seconds or so (or maybe when moving the board? Not sure)

What can I do?

DaneEvans, 2021-08-01:
It sounds like it's doing exactly what you've told it to.

It has the layer indicators turned on, and they take priority. If you
remove those that test pattern should work.

Geobert, 2021-08-02:
Thank you for your help! After removing layer related code, it almost work!

on master part, led 2 and 10 don't show red and plugin slave part on usb led 3 doesn't show red (only 2, 3 and 4 are soldered so far)

Maybe I should precise that the board is only with per key LEDs, (1-2 on j6 and j4)

DaneEvans, 2021-08-02:
What happens if you plug the other one in as master - does it follow the
board, or follow the master?

Geobert, 2021-08-02:
Sorry, I wasn't clear enough:
- left side alone on USB, led 2 and 10 don't show red (when following the labels from the pic linked in OP)
- right side alone on USB (which has only 2, 3 and 4 soldered), led 3 doesn't show red, two others are working fine

Geobert, 2021-08-02:
After replacing led 10 on left side, it works, so maybe I'm so bad that I overheated the led 2 each time, let's try again...

Geobert, 2021-08-02:
It was dead for red, replacing again worked, so I think I only have hardware issue now, thanks for pointing on the layer thing in the fw!

closed 2021-08-02 by Geobert