josefadamcik/SofleKeyboard #109: Vial support and encoder programming is merged!

kirbysayshi, 2021-08-29:
I just added Vial support for Sofle! Vial is very new to me, but it is similar to VIA in that it can live-program the keymap. Most importantly though, it supports live programming the encoders!

PR: https://github.com/vial-kb/vial-qmk/pull/50

This is what the UI looks like. You get separate up/down encoder "keys" that can be any QMK key.
[image]

Apologies for the noise, but I wasn't sure the best place to share

Thanks to Josef Adamčík and the various contributors for making a great keyboard! I use it everyday.

josefadamcik, 2021-08-30:
Hi and thanks a lot for the contribution. I haven't heard about vial until now but it looks nice and the support for encoders is indeed very useful. It's also very cool you went ahead and added the support to it. Thanks!

I am closing this issue since you already delivered the enhancement but I am happy you let us know. People can find the closed issue when they search for "vial" or even "via" and this will lead the to a solution and that's certainly good.

closed 2021-08-30 by josefadamcik

kirbysayshi, 2021-08-30:
Perfect, that was exactly what I was hoping!