josefadamcik/SofleKeyboard #110: Help: summarise your issue here.

HybridFox, 2021-09-05:
**Kind of issue?**
Help

**Which Board?**
Original

**Which Board Revision?**
1.0

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
So this is my very first DIY split kb, i have gotten everything working except the OLED.
I'm using an Elite-C v4 with the SSD1306 OLED, I bridged the jumpers and _think_ i have soldered everything correctly, but i just can't get the OLEDs screen to show anything on neither of the keyboards. The only thing i can see after enabling logging is the following: `oled_init cmd set 1 failed`

I have tried to search for the error, to see if anyone else has encountered this, but to no avail i only find the source pages of QMK.

Any help pushing me in the right direction is greatly appreciated!
Thank you so much to anyone who can help me in advance!

**What you expected:**
OLED screens working

josefadamcik, 2021-09-06:
Hi HybridFox,

The fact you are using Elite-C shouldn't matter for Sofle. I haven't tested it but it's a drop in replacement and the only problems I have heard about is the halve detection problem in one of the older revisions and the fact that RAW pin is connected differently on Elite-C. But the RAW pin is dis-connected on Sofle so that couldn't cause any issues.

Pro tip: You can also google for problems for Lilly58, Corne and other keyboards split keyboards where the connection to OLED is almost identical so the debugging and solutions will be the same. 

Basic Tips:

I would check with continuity meter / multi-meter if there's really a good connection between GND, VCC and the SDA/SCL i2c pins on the EliteC and and the OLED when everything in place (but not powered).

They are usually labeled on the OLED PCB, you'll have to look up the corresponding pins on micro-controllers in the schematic. I have attached pdf so you don't have to go to the KiCad. but you still have to check which pins are they actually on micro-controller.

[SofleKeyboard.pdf](https://github.com/josefadamcik/SofleKeyboard/files/7114338/SofleKeyboard.pdf)

Also make sure you you soldered other things correctly. E.g. one way how to break OLED is to modify the default setup for connection of the halves from serial to I2C. This is not even mentioned in the build guide (it's not recommended), so I assume you didn't try to do that for your first split build. 

Make sure you have the official default firmware with the default keymap from QMK and not using VIA and similar tools. They usually don't support OLEDs and Encoders as far as I know.

There are also the basic things: make sure the OLEDs actually work, double check all the connections, try it only with one half plugged in.

HybridFox, 2021-09-06:
Hey Josef, thank you for giving me some tips to look for the issue!

I have checked the GND & VNC continuity and they seem fine. Same with the SDA/SCL & pins on the EliteC
Though it seems i also have continuity between GNC & VCC itself, not sure if that's supposed to be like that? I'm finding conflicting info about that online.

I have indeed always flashed the keyboard with the default keymap using `qmk fash`, so that should be.

Do you know a way how i can test if the OLED is actually working?

Thanks again!

DaneEvans, 2021-09-06:
Try measuring the other way between GND and VCC, or do so without the micro
attached.  There definitely shouldn't be a short, but you are probably
seeing a diode between them.

Alternatively power up the board and see if there is 5v between them.

The easiest way to stop the oleds from working would be to do the jumpers
on the wrong side, or do both sets.  Have you checked that?

If you have spare oleds you could try another one.
Make sure there are no address or enable jumpers on the OLED, they aren't
in the basic design, but that's not to say that some manufacturer hasn't
decided it's a good idea.

As to the error. An init failed suggests to me a NACK error, which points
to a connection bug.

If you post some pictures we can maybe see something

jeremiah-miller, 2023-01-26:
I'm having this problem as well, using serial for communication between the halves. I don't even get the oled_init warning. I have the v1.1 pcb, what do you mean by having the jumpers on the wrong side? are the pcbs not reversible? I connected them after soldering in the elite-c.

Also, what do you mean by "Make sure there are no address or enable jumpers on the OLED"?
[image]

DaneEvans, 2023-01-26:
See, with an image we can help.

Read the build guide, and pay attention to which side of the board the OLED
jumpers are supposed to be on. And where the micro is meant to be.

I think you'll find that the designs are for the micro, and jumpers on the
top side of the board.

And your micro is on the top, but you soldered the jumpers on the bottom.
Hence the polarity and data lines are all reversed, and you probably fried
the OLED

jeremiah-miller, 2023-01-26:
That's what i was worried about, i missed the part about soldering the jumpers until after the micro was soldered in. 

Given that, if I got a new oled and desoldered/removed the old one, do you think it would work if i put the new one on spun 180 degrees? Or could i use wires to connect it in a way that the oled would work?

DaneEvans, 2023-01-28:
If you use the jumper pads to swap the order of them - ie 1>4, 2>3, 3>2,
4>1, or reverse the OLED, both should work.  The first would be cleaner.

jeremiah-miller, 2023-02-01:
I flipped them and the OLED still works! The pimoroni trackball I had connected works too, neither seem to have been fried! Thanks for the help!

closed 2025-06-18 by DaneEvans