josefadamcik/SofleKeyboard #111: RGB color on LHS does not match RHS

petertriho, 2021-09-09:
**Kind of issue?**
Assembly

**Which Board?**
RGB

**Which Board Revision?**
RGB =   2.1

**What steps have you taken**
* [X] I have searched the Issues page
* [X] I have read the documentation 
* [X] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
LEDs were working fine for a week now but suddenly the LHS LEDs have incorrect brightness and color compared to RHS

Video illustrating the issue with the LHS https://imgur.com/a/PPqbXLh

**What you expected:**

Color of RGBS should be the same on both sides

petertriho, 2021-09-09:
Nevermind, it's back to normal now after replugging the trrs cable

DaneEvans, 2021-09-09:
One of the early leds in the chain has a dry solder joint and is firing randomly. This is then passed through all of the others as a disco effect.

closed 2021-09-09 by DaneEvans

petertriho, 2021-09-09:
I tried looking at the solder joints, would that be the LED on the top right of this photo? https://imgur.com/a/GKuhi8T It seems to work but with a different color at different brightness levels