josefadamcik/SofleKeyboard #116: Encoder triggers adjacent switches

aadcg, 2021-10-17:
When I rotate the left encoder, the 4 adjacent switches are triggered (on QWERTY those are "5tgb").

I'm wondering if anyone has seen anything like this before.   

This is not firmware related (connecting the USB-C on the other half solves the issue).

Thank you.

DaneEvans, 2021-10-17:
You have a dry connection on pin 13 of your micro. 

The Col 5 pin is the common element for all of those keys, although it goes to the push button on the encoder not the rotation element. 

I'm surprised that you are getting false positives, but unless you have multiple shorts coincidentally on those, there's no other option. 

Does the rest of the keypad work normally? I'm really confused as to how you get that result. Not working I could understand, but false hits is bizzare

I'd resolder pin 13, then try replacing the encoder, but really I'd need to probe the board I think

aadcg, 2021-10-17:
I was able to solve the issue with the help of a friend. It turned out that I had to unbend some encoder pins so that they wouldn't touch the encoder itself.

[image]

Why did it work? I have no idea, but I'm happy. Does this make any sense to you @DaneEvans?

DaneEvans, 2021-10-17:
That makes some sense. 

You're getting some kind of short across the encoder footprint. Likely the ENC side to the switch side. 
The top of the switch is on that column. 

Again I would have thought that you'd get dead switches, given that the switch and a diode is in there before you get to where the measurement is taken, but maybe there's some noise or something. 

I'm glad you figured it out anyway