josefadamcik/SofleKeyboard #119: Keyboard Entry for remap-keys.app

Silverquark, 2021-11-02:
I've been using remap-keys.app for some time to quickly configure my corne keyboards layout.
Sadly there is no keyboard entry for the sofle and only the original author can add it. 
Setting is up is pretty straight forward and shouldn't take longer than a few minutes.

[Link to Remap](https://remap-keys.app/)

josefadamcik, 2021-11-14:
I have submitted it for review, we'll see how it goes. Thae do require some json with layout which I don't really have time or energy to study and create (it's certainly not about few minutes). Hopefully the one that was used for Via would be enough.

josefadamcik, 2021-11-15:
So it was approved and I added some information. Hope that does the job for you, I haven't tested anything ( I have no clue what the service is about).

closed 2021-11-15 by josefadamcik