josefadamcik/SofleKeyboard #124: Help: (Choc) split RGB LED and OLED issues

blewisemt, 2021-11-22:
**Kind of issue?**
I cant seem to get the RGB matrix pattern to change on the slave side of the board. I also can't seem to get the OLEDs to function at all or get the board to change layers. The board defaults the "breathing" pattern, and is on the first layer.

**Which Board?**
Choc  

**Which Board Revision?**
2.1

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing? Yes, all of them

**What is the issue?**
I initially used the Choc firmware at https://github.com/brianlow/qmk_firmware, built/flashed for VIA per the Choc build guide, resulting in the innermost vertical column of LEDs on each side working, and the OLEDs functioning. I then reflashed using the firmware at https://github.com/brianlow/qmk_firmware/blob/5264ae0ce57033322638b87ada26a377a4b25eab/keyboards/sofle/keymaps/choc-brian/config.h#L17-L19. This resulted in all of the LEDs working, but the slave half is stuck in the breathing matrix pattern. Additionally, The OLED screens no longer work. Perhaps I don't know how the keyboard works, but I can,t get the board to switch layers.

**What you expected:**
I'm trying to change RGB modes to either solid or heatmap. I would also like the OLEDs to indicate the layer, and to be able to be able to swap layers. (How is this normally done?)

brianlow, 2021-11-22:
Thanks for opening a new issue.

> This resulted in all of the LEDs working, but the slave half is stuck in the breathing matrix pattern.

I have the same problem. If you don’t change lighting much, a workaround is to plug just the right side in to the computer, change the lighting then plug everything back normally. You'll need temporarily map some keys on the master half to adjust lighting. If that isn't acceptable, you'll need to use the regular `choc` firmware. Last I looked VIA didn't seem to be getting much love. Be interesting to see if Vial is a good option.

> swap layers. (How is this normally done?)

To switch to a layer, hold the raise or lower key (blue and red here):
[image]

Pressing both raise + lower enters an "adjust" layer where some keys will change things like lighting. Since you are using the `choc_brian` firmware, download and run https://caniusevia.com/ and it will let you view/change your keymap on the fly.

[image]

>  I would also like the OLEDs to indicate the layer

Perhaps flash the regular choc firmware to double-check the OLEDs wiring is still good - maybe a marginal connection. Next, I'd compare the OLED code in the two firmwares. I moved it around between versions but check out `choc/keymap.c` (at the bottom) and `choc-brian/oled.c`. Maybe change the firmware to just print a fixed string "Hello World" to see it can talk to the device.

blewisemt, 2021-11-22:
So I think I've solved the layer issue using the information you provided. Still working on the RGB, as the first firmware I tried replicated the initial issue. OLEDs worked again as well. Transitioned back to the firmware you suggested, and once again nothing on the OLEDs.

brianlow, 2021-11-22:
Some ideas for the OLED issue:
* simplify the `choc_brian` firmware to just single print or display the logo
* change the `choc_brian` firmware to more closely match the working `choc` firmware
* ensure flashing same firmware on both sides
* try getting one OLED working with just one half connected

brianlow, 2021-12-23:
@blewisemt just seeing if you had any luck solving

antoineroux, 2022-01-24:
I just want to report I had the same issue with the right side not changing animation or color. The workaround worked for me.

Is it more a QMK bug or an issue with the keymap?

derpus-potatus, 2022-01-30:
I believe I managed to solve the RGB issues. I am able to change animations and toggle the LEDs on/off on both sides using keys on either side of the board, no need for the workaround described earlier in the thread. My OLEDs also work fine. However... a preface: 

- I don't work with code but do know juuuust enough to get in trouble. QMK and a bit of ZMK is the extent of my "experience".
- I have no idea how to use git/github effectively so my changes are just written out below. 
- I did not fork brian's qmk repo nor use the choc branch as directed in the build guide. Therefore, I'm probably working with a "newer" version of QMK branch; as a result there may be other little differences that I forgot to capture below that may prevent you from compiling the firmware. Apologies.

With that out of the way, I suspected that the two sides were not communicating certain instructions. Given that normal modifiers (shift etc) work, there must be another function that communicates things like power, RGB effects etc. 

After managing to get the two sides to accept commands like changing RGB modes, the RGB on the slave side no longer illuminated on their own (did not follow their defined "RGB_MATRIX_STARTUP_MODE"). 

At that point, I guessed that the driver/controller did not know how to address the RGBs, or that the RGBs existed. So I increased the number of LEDs in config.h and also "defined the matrix" for the slave side (this was very difficult for my weak untrained mind). 

I've put my keymap/config/rules files here: https://github.com/derpus-potatus/sofle-choc/blob/main/config.h

And here are the changes that I remember making:

**In config.h**

added:
```
#define SPLIT_TRANSPORT_MIRROR
```

changed lines 18 and 19: 
```
#define RGB_MATRIX_SPLIT { 29, 29 }
#define DRIVER_LED_TOTAL 58
```

**In keymap.c**

In the block of code where the RGB matrices are defined, I added definitions(?) for the right side (master is left). The whole block now reads:

```
// RGB Matrix Configuration
#ifdef RGB_MATRIX_ENABLE

led_config_t g_led_config = { {
  // Key Matrix to LED Index
    // Left
    { 28, 21, 20, 11, 10,      0 },
    { 27, 22, 19, 12,  9,      1 },
    { 26, 23, 18, 13,  8,      2 },
    { 25, 24, 17, 14,  7,      3 },
    { 16, 15,  6,  5,  4, NO_LED },

    // Right
    { 57, 50, 49, 40, 39,     29 },
    { 56, 51, 48, 41, 38,     30 },
    { 55, 52, 47, 42, 37,     31 },
    { 54, 53, 46, 43, 36,     32 },
    { 45, 44, 35, 34, 33, NO_LED }

}, {
  // LED Index to Physical Position
    // LEFT
    {  95,   7 }, {  95,  21 }, {  95,  36 }, {  95,  50 }, { 115,  64 },
    {  95,  64 }, {  76,  62 }, {  76,  48 }, {  76,  33 }, {  76,  19 },
    {  76,   5 }, {  57,   3 }, {  57,  17 }, {  57,  31 }, {  57,  46 },
    {  57,  62 }, {  38,  64 }, {  38,  48 }, {  38,  33 }, {  38,  19 },
    {  38,   5 }, {  19,   7 }, {  19,  21 }, {  19,  36 }, {  19,  50 },
    {   0,  50 }, {   0,  36 }, {   0,  21 }, {   0,   7 },

    // RIGHT
    { 129,   7 }, { 129,  21 }, { 129,  36 }, { 129,  50 }, { 119, 64 },
    { 129,  64 }, { 148,  62 }, { 148,  48 }, { 148,  33 }, { 148, 19 },
    { 148,   5 }, { 167,   3 }, { 167,  17 }, { 167,  31 }, { 167, 46 },
    { 167,  62 }, { 186,  64 }, { 186,  48 }, { 186,  33 }, { 186, 19 },
    { 186,   5 }, { 209,   7 }, { 209,  21 }, { 209,  36 }, { 209, 50 },
    { 224,  50 }, { 224,  36 }, { 224,  21 }, { 224,   7 }

}, {
  // LED Index to Flag
    // LEFT
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1,

    // RIGHT
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1, 1,
    1, 1, 1, 1

} };

#endif
```

brianlow, 2022-01-30:
@derpus-potatus thanks for posting! I should try getting my branch on latest QMK

killua99, 2022-03-31:
@derpus-potatus this somehow could / might work with ZMK firmware ?

Lizardbutt, 2024-10-16:
Chiming in - in `rules.mk` you **must** set `RGB_MATRIX_DRIVER` to **exactly** `ws2812` as opposed to what is there now: `WS2812` I've been beating my head against the wall for 4 hours on this setup. It's 3:38AM and I have work at 8AM.

Lizardbutt, 2024-10-18:
> @derpus-potatus thanks for posting! I should try getting my branch on latest QMK

When?