josefadamcik/SofleKeyboard #126: Help: some random keys not working

antoineroux, 2021-12-22:
**Kind of issue?**
Assembly 
**Which Board?**
Choc   

**Which Board Revision?**
Choc = 2.1

**What steps have you taken**
* [X ] I have searched the Issues page
* [X ] I have read the documentation 
* [N/A ] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
Hello,
First of all, thanks everyone contributing to the project, this is the first time I build a keyboard, and it is fun.

I am building a choc and did the first half. I soldered the diodes, the sockets and a few LEDs (not all of them yet, I wanted to test first). I tested all the keys and a few random ones are not working:
* Row 0: `1` - `2` - `3` - `5`.
* Row 1: `T`.
* Row 2: `Tab`.
* Row 3: `Alt`.

`3` and `5` and `T` seem to be caused by faulty diodes, because only one side of the diode works.

I focused on the `1` key on row 0 next. I suspect if I can fix this one, the solution might apply to other keys as well. But this is where I don't understand what is happening. I can reproduce the same issue on `2`

When I connect both sides of the sockets with a piece of wire for any working key, it outputs the corresponding letter/symbol, as expected. When I do the same on the `1` key, it does nothing.

When I connect the right side of the socket of the `1` key to the left side of any working socket on the same row (or any diode on a working socket), it outputs the 1 correctly. I can reproduce this test on both sides of the board. If I connect the right side to its own diode, it works fine as well.

I removed the socket, at some point I was wondering whether it could be damaged, but that changed nothing. I can output the character by connecting the right side of `1` socket to any working diode on the same row.

So to me it looks like the issue is on the PCB:
* When input to `1` is sent anywhere else, it works.
* When input to a working key is sent to `1` output, it doesn't work.
* When input to a working key is sent to `1` diode, it works fine.

So the input is there, the microcontroller is correctly connected, this diode is working and the output does not seem to get to the diode.

Am I correct? Is there anything else I should check? What do I do if it is the board that's damaged? Remove the components and start a new one?

For completeness, other issues I have on the board:
* I soldered 6 LEDs, one of them is not working. I don't worry about that for now, I just need to replace one or two of them.
* The OLED is not working. This could be my fault: at first I bridged the wrong side of the board. I wasn't sure what to do, so I removed the solder on that side and bridged the other side, except I had to do it under the microcontroller that was already soldered in, so I might have failed.

**What you expected:**

I guess I was expecting that to be my fault, with a faulty diode for instance.

Thanks in advance.

DaneEvans, 2021-12-22:
There are only 4 things in those switch circuits

Micro output (row I think, but check it)
Switch
Diode
Micro input(col?)

For a failure of a whole row or column, check the micro sockets.
There are also ways for the diodes to screw these up, but I'd need to look
at a schematic

For a failure of a single switch, then look to the switch (which you have)
or the diode.

First things first I'd check that all your diodes are around the right way
(hopefully on that board they are all the same orientation)

After that I'd check the schematics and see if a circuit row or column
contains most or all of those switches. The circuit may not reflect what
the key matrix looks like it should be.

antoineroux, 2021-12-22:
Thanks a lot for the answer. 

So in my case, there isn't a full row or column that isn't working. I have two failing keys on the same row, but that's it (and the other two are caused by bad diodes). So I believe it rules out the micro-controller. Also, I can get it to output the non-working keys by connecting the input to any working output or directly to the diode. So I think it rules out the diode.

To clarify a bit, here are pictures of what doesn't work (in red) and what does work (in green). Imagine the lines being a piece of wire.

[image]
[image]

I used solder paste, I did my best to remove it when I desoldered. Coult it be it?

brianlow, 2021-12-22:
Some ideas...

> 3 and 5 and T seem to be caused by faulty diodes, because only one side of the diode works.

Replace the bad diodes first. I think it is good to solve the known problems even if not obvious how they may affect the other keys.

Double-check diode orientation on all keys

Test the diodes in-place if your multi-meter supports it. Should see 0.5V with red probe on the bottom diode lead and black probe on top lead. Then should see nothing (0L) when probes are reversed.

Check continuity for the lines below. For example, the red lines below indicate the right side of the switch sockets should be connected together.

[image]

antoineroux, 2022-01-07:
Thanks a lot, that really helped. I started by applying your first advice and replaced the diodes I knew weren't working. Then I realised while testing the diodes for the key `1` that's causing me trouble, I was touching the diode's leg: that proves that the diode is working, not that it is correctly soldered.

I replaced the diode and now it is working. I still have a few keys to fix, but I feel like I know enough to do the debugging. I will close this issue. I will open a new one if I find a new problem.

Thanks a lot for your help.

closed 2022-01-07 by antoineroux

antoineroux, 2022-01-12:
Just a follow-up: I fully built one half, so I expect the second half won't be an issue.

brianlow, 2022-01-12:
Great to hear. Thank for the update. Let us know how the full build goes

antoineroux, 2022-01-24:
Here is the keyboard, completed. 

It was the first time I was building my own keyboard, so I remained very close to the original design. Sourcing all the parts and soldering everything was already a lot of learning. 

I am now tweaking the firmware. 

Thanks again for publishing this keyboard and foe the help. 

[image]

[image]

antoineroux, 2022-01-24:
The other pictures did not make it through. Here they are. 

[image]

[image]