josefadamcik/SofleKeyboard #129: Enhance: JLCPCB suggestion

STotev, 2021-12-23:
Hi. I put my PCB order for Sofle RGB and got the following email from JLCPCB:

> Hi Sir or madam,
> 
> Well got your order with many thanks~
> 
> Sorry to bother you, but there is an issue that we want to confirm with you before proceeding.
> 
> As shown below, the spacing between slots and outline is less than 4mm, too narrow, the board will break easily and missing during production. Can you pls amend the file by increasing the spacing to 4mm and replace new file?
> 
> Or just keep it as it is,but we won't accept quality compliant if board breaking cases happen, please kindly consider it carefully, tks.
>
>  [image]
>
> Your early reply will be highly appreciated!
>  
> Thank you so much!

I haven't seen anyone complaining about broken plates of this and according to the guide, the same manufacturer was used. Though, I wanted to share my experience and let you know about this in case someone wants to fix it.

dmikhalsky, 2021-12-29:
I recently got a Sofle RGB PCB set (top/bottom/main) from JLCPCB. No issues. 
The quality is fantastic btw. It's not my first time ordering from them, but these black-masked Sofle PCBs are a piece of art

STotev, 2021-12-29:
@dmikhalsky I just received an email that my order was shipped. They told me they'd let me know if any of the pieces failed but that did not happen so I guess everything was okay.

Out of curiousity - did they charge you extra £11 ($15), after you put your order, because of the top piece?
After letting me know about the issue above, they've sent me the following:
> And there are too many small slots in your file , it will take too much time to get it finished during the routing process. So there is extra cost $15.2 for it.

I just want to compare my experience with others as this is my first time ordering from them (or PCBs in general).

dmikhalsky, 2021-12-29:
No, they didn't. Sorry, man.
Actually, JLCPCB Customer Service is very dependant on the particular manager that deals with your order. This is especially true in case of combined PCBs. One might say that they are different PCBs and take a surcharge, another one might just leave it as is.

DaneEvans, 2021-12-29:
I think I got warnings from the automated tool, which I ignored.

It's a hobby board, I don't need it to last 10 years and have 100% yield.

Definitely no extra charges.

I'll look at the assembly in a few weeks when I get home, and see if the
dozen or so OLEDs I have are consistent enough to make those parts a few mm
wider

STotev, 2021-12-29:
@dmikhalsky thank you. That was helpful.

@DaneEvans thanks. 

I guess I was a special case or just other people couldn't bother reporting such warnings. I guess the board itself is okay and my order was just been held by a manager who really cares about good feedback. Which is a good thing. It definitely made a good impression about their services.

brianlow, 2022-01-01:
For the Sofle Choc, I've been charged the $15 USD routing fee and gotten the manual too-thin warning. 

Fee seems to be a flat fee per order, per color (or per PCB attributes). This is based on two separate orders, second order containing several dozen plates, both white and black. 

The Choc plate is nearly identical to the RGB plate (Choc is fork of RGB).

STotev, 2022-01-13:
Just received mine. Excellent job done by JLC. Nothing broken and the plates look very sturdy so it doesn't look like they can break easy but I guess they had to warn me.

dtrinf, 2022-03-06:
I just send mine today, and they told me I have to pay extra, really unhappy with JLC. 

```
An additional charge is applied to Y7-3815581A
2022-03-06 16:50:58

You need to pay additional €13.92 for your order.

1.There are too many small slots in your file , it will take too much time to get it finished during the routing process.
```

Isaquelc, 2022-04-17:
I also recieved this warning for the top plate of the Choc version, and a notice that I need to pay 15.2USD extra. The notice was this: 

> "the spacing between slots and slots to outline is less than 4mm, too narrow, the board will break easily and missing during production. Can you pls amend the file by increasing the spacing to 4mm and replace new file?
> 
> Or just keep it as it is,but we won't accept quality compliant if board breaking cases happen, please kindly consider it carefully, tks."

has anywone tried to thicken these edges or would it cause a problem with the placement of the microprocessor. Also, is there any reason the top plate is two layered? I tried ordering it in aluminium instead of FR-4 but they don't do more tha 1 layer for the board at the moment. Can I just pick any layer?

josefadamcik, 2022-04-19:
@Isaquelc  
> Also, is there any reason the top plate is two layered? I tried ordering it in aluminium instead of FR-4 but they don't do more tha 1 layer for the board at the moment. Can I just pick any layer?

The case plates don't need to have any copper layer at all, it's not functional. So you can throw them away.

dmikhalsky, 2022-04-19:
> @Isaquelc
>
>
> The case plates don't need to have any copper layer at all, it's not functional. So you can throw them away.

But you lose lizards :-)))

MrFoxPro, 2023-01-17:
Any workarounds so far?

josefadamcik, 2023-01-17:
@MrFoxPro 
> Any workarounds so far?

Make the plates from laser cut acrylic or whatever other material.

Honestly, it might be the case that JLCPCB wasn't consistent in this matter ,and most of the time they just make the stuff without problems. But I don't  really have any way how to tell for sure. I have never got any problems with any of the boards I made and there was only a handful of people who came back with an issue. At one point I had a person having the problem and at the same time other people saying it went through without issues.

Of course, JLCPCB are evolving their process so things change all the time. And I haven't made a new board in a long time, so take everything I say with a grain of salt.

MrFoxPro, 2023-01-17:
> @MrFoxPro
>
>
> Make the plates from laser cut acrylic or whatever other material.
>
> Honestly, it might be the case that JLCPCB wasn't consistent in this matter ,and most of the time they just make the [...]

Yeah thanks for you reply and great work on project...
I would like to print plates or make it from wood, however, I don't really much into DIY/solid world and have no idea how I can use Kicad project or SVGs for 3d printing/CNC cutting.

MrFoxPro, 2023-01-17:
Can I just use https://github.com/josefadamcik/SofleKeyboard/blob/master/Design/v2/top_plate.svg as blueprint, make 3D model in Blender and order for printing?

MrFoxPro, 2023-02-04:
This details were printed with ABS
[image]