josefadamcik/SofleKeyboard #134: Help: summarise your issue here.

alecharp, 2022-01-31:
**Kind of issue?**
_Bug / Assembly / Software / HELP!_

It's kind of all 4..

**Which Board?**
_Choc _  

Not from keyhive, directly ordered the PCB at JLCPCB with the gerber files on the repository.

**Which Board Revision?**
Choc = 2.1

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [ ] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**

I'm using two KB2040 from Adafruit and trying to have KMK (with circuitpython). I don't think this is related to the firmware but to the board. 
The pins to to UART Rx cannot be used to do Tx (I'm a noob in electronics, sorry), or it seems from https://github.com/adafruit/circuitpython/issues/4502. 

This would means that the fact the LED are connected to the Tx (D0) and the Data to the Rx (D1) is problematic. For the right half, the lines should / needs to be inverted. What do you think?

Also, the current pcb is made to work in one direction. Did I understood that correctly? Does this means that the two half cannot sync up? Meaning a layer toogle change on the left side couldn't trigger a layer change on the right? 

If the LED line was connected to another pin, we could have Rx/Tx pins linked to the TRRS connector and have the firmware handle the Tx/Rx cross connection. Did I understand that correctly as well?

KB2040 pinout: https://cdn-learn.adafruit.com/assets/assets/000/106/984/original/adafruit_products_Adafruit_KB2040_Pinout.png?1638564074

**What you expected:**

Nothing really. I mostly have questions and would like to know if I can "fix" the PCB I have, or if I have to spend time on changing the PCB design and order new ones.

alecharp, 2022-01-31:
Side note, my point on the `Tx/Rx` is also based on https://github.com/KMKfw/kmk_firmware/issues/198#issue-872977177.

brianlow, 2022-01-31:
This is cool. I think the RP2040 might become a really popular keyboard controller. Some first thoughts:

> For the right half, the lines should / needs to be inverted

I don't think so if the controllers are installed in the holes with the outline around them. If so the controllers will be wired identically

brianlow, 2022-01-31:
> Also, the current pcb is made to work in one direction. Did I understood that correctly? Does this means that the two half cannot sync up? 

With ProMicros the two halves work independently and then cooperate using the TRRS cable. So changing the layer with a keypress on the left will also change the layer on the right half

brianlow, 2022-01-31:
I've read the RP2040 has a feature called programmable I/O so I think most pins can be configured to control the LEDs

alecharp, 2022-01-31:
> I don't think so if the controllers are installed in the holes with the outline around them. If so the controllers will be wired identically

yes, the D1 is connected to the other half D1. 
But in CircuitPython, I cannot create a UART connection with `busio.UART(tx=board.RX, rx=None)`. The code won't load.

In the end, it's probably a problem of CircuitPython and not Sofle design. I'll more testing.

alecharp, 2022-01-31:
Based on https://github.com/adafruit/circuitpython/issues/5953 response, we can close this. 

Even if it would be best to have the Tx/Rx pins to cross, it possible to get around that. I _just_ need to work on KMK split module to use pio and not the board UART.

closed 2022-01-31 by alecharp

brianlow, 2022-01-31:
Good luck, let us know if you solve it!

alecharp, 2022-02-01:
as I'm not good in electronic, and I mostly code in Java, not Python even less CircuitPython, that will be a challenge.

netgmx, 2022-02-06:
I'm working on the redox keyboard drived by two pi pico, here is what I found:

- pico doesn't like switching rx and tx pins, I needed to cross wires between halves on uart0 (pins 0 and 1 ). There is not used flag uart_flip - maby this was it intention but not implemented?

- there is somewhere in libs bug (or in my code but it's minimal), killing script if there is two way communication between boards so i had to change split.py module between halves

left side (usb)
self._uart = busio.UART(rx=self.data_pin)

right side
self._uart = busio.UART(tx=self.data_pin2)

not nicest solution and now that I know what is going on I can design some logic around it, hope it helps someone

Tonasz, 2022-02-07:
I accidentally found this issue as it was linked to KMK repository. I got some news which could interest you @alecharp:
- The https://github.com/KMKfw/kmk_firmware/pull/301 PR was merged and fixed some issues with wire communication (for example it fixed `uart_flip` you @netgmx mentioned)
- I have similar issue as you (but on Kyria board, also one pin TRRS connection) and found similar tips about using PIO. I bashed together the examples from documentation and my board worked. Made PR for you to look at: https://github.com/KMKfw/kmk_firmware/pull/320 Please mind that this will be only one way communication on one wire
- "Also, the current pcb is made to work in one direction." - I don't know choc, but QMK has one-wire, custom bitbanging protocol implemented, so two-way communication works on one pin. KMK doesn't have it implemented yet (nor circuit python I guess).