josefadamcik/SofleKeyboard #135: Enhance: enable GitHub discussions?

brianlow, 2022-02-04:
Wondering if would be good to have a place to discuss Sofle keyboards on topics that don't fit the format or formality of a GitHub issue. Another idea is a Discord server since Discord is common in the community. Willing to help

josefadamcik, 2022-02-04:
I honestly didn't even notice github had discussions :D 
I have no issue with turning them on and trying out. I am having e-mail exchanges with people about Sofle here and there and I could engage in the discussions in similar manner. The pro would be more people can read it later.  So far I like the idea.

That said, I wouldn't have the capacity to engage in moderation and maintenance, though. That's also one of the reasons I didn't want to start a Discord server. From my past experience those things always need more time than anticipated.

But I am happy to try it, maybe I would prefer to try the discussions here and see how it goes. The worst case would be I turn it off again if it becomes toxic or something. 

Or do you think (and I am asking others as well) that Discord would be so significantly better we should try that out?

brianlow, 2022-02-05:
Good point on the moderation time. Hadn't considered that. 

I'm up for trying discussions with the idea that the bar is low for turning it off.

josefadamcik, 2022-06-01:
I completely forgot about this one! I have just enabled discussions. We'll see how it goes.

closed 2022-06-01 by josefadamcik