josefadamcik/SofleKeyboard #136: Sofle Choc: Should the top plate thickness be 1.6mm or 1.2mm

infused-kim, 2022-02-14:
**Which Board?**
Choc

**Which Board Revision?**
2.1.1

**What is the issue?**
The [build instructions for the Sofle Choc](https://josefadamcik.github.io/SofleKeyboard/build_guide_choc.html) don't specify the thickness of the top plate and instead link to the [general sourcing page.](https://josefadamcik.github.io/SofleKeyboard/sourcing_parts.html), which specifies a thickness of 1.6mm for both the PCB and the top plate.

But it seems like the required board thickness for the Kailh chock switches is 1.3mm as per [Kailh's own schematics](https://github.com/keyboardio/keyswitch_documentation/blob/master/datasheets/Kailh/CPG135001D02rev1_Brown.pdf):
[image]

This is also the recommended thickness by [Keebio](https://docs.keeb.io/choc-stabs):
> If you're designing your own switch plate for Choc switches, **your plate must be 1.2mm thick so that the Choc stabilizers can clip into the plate properly**. The stabilizers will not clip into thicker plates properly.

You can also see a plate design here that is using 1.3mm:
https://imgur.com/a/Rv9xgRU

I selected `HW Bug` as the issue type, but I am not sure if it actually is. 

I want to stress that this is my first ever build. I have never even touched a mechanical keyboard switch... So maybe a 1.6mm plate is just fine.

Keebio stressed that chock stabilizers won't clip into thick plates properly. But what about the switches themselves?

Are the little plastic clips flexible enough to clip into a 1.6mm plate?

**What is your experience with using a 1.6mm plate?**
Those of you who have built a sofle choc @brianlow , @lytedev, @blewisemt... Did you order a 1.6mm or a 1.2mm top plate and did the choc switches properly click into your boards?

infused-kim, 2022-02-14:
Fantastic, thank you. 

I placed an order for my 1.2mm PCB today before seeing your reply. I was too impatient ;)

But at least we will know if 1.2mm work too!

lytedev, 2022-02-17:
I ordered 1.2mm plates and they worked great. I googled for Kailh Choc schematics which indicated that the gap is 1.3mm on the switch itself:

[image]

brianlow, 2022-02-17:
@infused-kim can you re-open?

I was surprised 4mm didn't matter so I I just double-checked.  I manufactured some with 1.2mm and some with 1.6mm. The switches snap into the 1.2mm plates but don't into the 1.6mm. 1.6mm still works - the switches are held in firmly by friction and usually you are pressing down anyways. But they are easier to remove from a 1.6mm plate than a 1.2mm because the lower catch doesn't engage. Should update the docs

infused-kim, 2022-02-17:
Ah very interesting. Good to know!

I have re-opened the issue.

infused-kim, 2022-02-19:
One more thing I noticed in the [Kailh Choc socket spec](https://www.kailhswitch.com/Content/upload/pdf/202115927/CPG135001S30-data-sheet.pdf?rnd=94):
[image]

So perhaps it would make sense to use a 1.2mm thickness for the PCB too.

brianlow, 2022-02-19:
oh interesting, the switch may be better seated, thanks for posting

johnnytshi, 2022-04-04:
@infused-kim with1.2mm, the bump will protrude the PCB (0.05mm), and keys won't sit flat.

on [Kailh website](http://www.kailh.com/en/Products/Ks/CS/319.html), under "Precautions":
2.Please use the thickness 1.6mm PCB

infused-kim, 2022-04-05:
> @infused-kim with1.2mm, the bump will protrude the PCB (0.05mm), and keys won't sit flat.
> 
> on [Kailh website](http://www.kailh.com/en/Products/Ks/CS/319.html), under "Precautions": 2.Please use the thickness 1.6mm PCB

Thank you so much for letting us know.

Sounds like 1.6mm for the PCB and 1.2mm for the top plate are the right numbers for choc keyboards.