josefadamcik/SofleKeyboard #139: Help: Bridged the OLED jumper pads on the wrong side of the PCB

nilpath, 2022-02-21:
**Kind of issue?**
Help!

**Which Board?**
Original

**Which Board Revision?**
1.0

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [ ] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**

Hi,

I'm sorry if this is not the right place to ask this kind of help but I would appreciate it a lot.

This is my second mechanical keyboard I'm building, and the first where I'm doing this much soldering 😅.

By misstake I bridged the the 4 OLED jumper pads on the bottom side of the PCB instead of the top side, Is this going to be a problem? 

If i should redo the soldering on the top side instead, is it fine to leave some solder on the pads on the bottom side as long as they are not connecting to each other?

josefadamcik, 2022-02-21:
Hi,
yes, it is an issue. You'll have to remove the bridges and bridge them on the correct side. But it's fine to leave the solder there as long as there's no electrical connection there (maybe check with multi-meter if you have one).

But from my experience it's much harder to actually bridge them, than to un-bridge them. So it should be easy.

nilpath, 2022-02-21:
> yes, it is an issue. You'll have to remove the bridges and bridge them on the correct side. But it's fine to leave the solder there as long as there's no electrical connection there (maybe check with multi-meter if you have one).

Roger!

> But from my experience it's much harder to actually bridge them, than to un-bridge them. So it should be easy.

Yeah I was able to quite easily remove the bridge but I have quite cheap equipment so it's a bit of a pain to remove all the solder

Thank your for the help! 🙂

closed 2022-02-21 by nilpath

DaneEvans, 2022-02-21:
To remove it fully, get some solder wick and some flux, and then place the
wick on the pad with the iron on top.  All the solder will be sucked into
the wick, and if you clean up the burnt flux with isopropyl you should be
pretty close to a new pad.

But yeah, it's not needed.  Just break the connections, and redo it on the
correct side.