josefadamcik/SofleKeyboard #141: SK6805/SK6803 as alternatives for per-key RGB

Sachiru, 2022-03-27:
**Kind of issue?**
 Assembly 

**Which Board?**
 RGB

**Which Board Revision?**
RGB =  2.0, 2.1

**What steps have you taken**
* [X] I have searched the Issues page
* [X] I have read the documentation 
* [X] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

Is it possible to substitute a SK6805 or SK6803 for the SK6812s to lower power draw a little if one wants to enable all the lights and is using something like a Proton C or Bonsai C4?

DaneEvans, 2022-03-27:
Not out of the box, but feel free to do up a board.
They are still 5v, so it should just be a footprint swap. But I can't
imagine that soldering LEDs that small will be much fun

Sachiru, 2022-03-27:
What would be a good alternative that is pin compatible but draws less power? SK6805?